mani-skill/ManiSkill #166: Unstable simulation in Pour-v0

pauljansonnie, 2023-11-01:
Hi,

The fluid simulation in Pour-v0 is unstable. In some runs, the fluid seems to explode/implode. This does not happen in every run but still very often. I noticed this by running `python -m mani_skill2.examples.demo_random_action -e Pour-v0 --render-mode=human`. Here is a short clip illustrating this issue:
[pour-v0_bug.webm](https://github.com/haosulab/ManiSkill2/assets/75021646/c9a63c7a-611e-4e18-9f94-015edcdce8b8)

Is this a known bug? Is there any way to prevent this from happening?
I checked the initial configuration of the fluid and it is basically multiple layers of particles on top of each other (similar to what is shown in #49). Is there a reason why you made this design choice as for the initialization of the fluid?

Thank you.

xuanlinli17, 2023-11-01:
Try changing to larger simulation frequency might alleviate the issue? Though you will have slower simulation

pauljansonnie, 2023-11-07:
It looks like using a larger `sim_freq` actually makes this worse. When using `sim_freq=2000`, I consistently get this behavior. When using `sim_freq=200`, it is less frequent. However, for a fixed `sim_freq`, augmenting `mpm_freq` seems to solve the issue. So, it is probably the ratio between these that was causing this issue. Thanks.

closed 2023-11-07 by pauljansonnie