mani-skill/ManiSkill #179: Force Torque sensor support

lauweiquan, 2023-12-16:
Hi, I would like to know if maniskill2 supports the use of force torque sensors and how do I implement it?

Thank you

xuanlinli17, 2023-12-17:
Hi, You can calculate the approximate torque using pd parameters
See https://github.com/haosulab/ManiSkill2/issues/144

lauweiquan, 2023-12-17:
thank you for the prompt reply!

lauweiquan, 2024-01-22:
@xuanlinli17 
Hi, I am also looking to calculate the force and torque value of the joints that mimics the values of a force torque sensor in real life. Will that be possible in Maniskill2? Thank you.

xuanlinli17, 2024-01-22:
@fbxiang @StoneT2000 

I think we plan to add this in ManiSkill3?

StoneT2000, 2024-02-02:
At the moment we do not have this. We will revisit this feature in the future most likely for CPU simulation at least. I will leave the issue open for now.

StoneT2000, 2024-05-23:
Some progress has been made on this front @lauweiquan https://github.com/haosulab/SAPIEN/discussions/156, it is not fully integrated yet into maniskill, needs to be tested

StoneT2000, 2025-08-10:
Based on what we have seen when using PhysX, it is better to disable gravity on a fixed-base robot entirely instead of using the GPU sim's gravity compensation. Regarding coriolis forces, maybe @fbxiang knows, but I don't know if we have the API on the GPU sim for getting those values.