mani-skill/ManiSkill #182: Wiki of all objects in maniskill

CyanWatts, 2023-12-19:
Is there a way to quickly browse all available objects in Maniskill?

xuanlinli17, 2023-12-19:
The objects vary between different environments and we don't currently support browsing objects in web. However, when you create an environment, you can pass `model_id=xxx` as an argument and render the environment to visualize what the object looks like in the environment.

StoneT2000, 2023-12-20:
This will be a project we may have someone work on soon. We would like to make something similar to https://vigorous-lamarr-a0a9c0.netlify.app/ which lets people browse through all of the ProcTHOR/AI2Thor objects (which we will support as part of v0.6.0)

CyanWatts, 2023-12-20:
> The objects vary between different environments and we don't currently support browsing objects in web. However, when you create an environment, you can set `model_ids` to a specific object id (you can get the object id from the json files under `data/`, and render the environment to visualize what the object looks like in the environment.

Thanks for the reply! Actually I knew this method, but it's a bit inconvenient. You can't get a quick look through ALL KINDS OF assets within a singe page/GUI.

CyanWatts, 2023-12-20:
> This will be a project we may have someone work on eventually. We would like to make something similar to https://vigorous-lamarr-a0a9c0.netlify.app/ which lets people browse through all of the ProcTHOR/AI2Thor objects (which we will support as part of v0.6.0)

Thanks for the reply! That's exactly what I mean. I'll keep an eye on this.