mani-skill/ManiSkill #194: [bug] Config error of xmate3_robotiq, resulting in abnormal rotation of rx, ry, rz !!!

swearos, 2024-01-25:
https://github.com/haosulab/ManiSkill2/blob/d04c2dc5bfd826172b2f3193cbf232a4a33bdd9e/mani_skill2/agents/configs/xmate3/defaults.py#L43

`self.ee_link_name = "grasp_convenient_link" ` should be changed to
`self.ee_link_name = "robotiq_arg2f_base_link"`,

I tried to rotate the end controller 90 degrees around the X-axis and noticed that the robotic arm was acting strangely,
See the video below for a comparison of the effect before and after modification (also problematic is the rotation of rz).

https://github.com/haosulab/ManiSkill2/assets/10892059/d5523cc7-7f9c-469d-9e24-676c4d9385f9

https://github.com/haosulab/ManiSkill2/assets/10892059/f1149dea-18be-4632-8aac-94d1533bb159

StoneT2000, 2024-01-26:
thanks for raising the issue @swearos, could you provide a reproducible code snippet to do this?

swearos, 2024-01-29:
@StoneT2000
Script has been attached, the default panda robot rotation results are normal, xmate3_robotiq rx and rz are abnormal, you can try it
 `
import torch
from tqdm import tqdm 
import gymnasium as gym

import os 

import numpy as np
np.set_printoptions(suppress=True, precision=4)
from mani_skill2.utils.sapien_utils import look_at
from mani_skill2.utils.registration import register_env

pose = look_at([-0.6, 0.4, 0.6], [-0.1, 0.2, 0])

control_mode="pd_ee_delta_pose"
env = gym.make(
        "PickCube", obs_mode="rgbd",render_camera_cfgs=dict(p=pose.p, q=pose.q), control_mode=control_mode, 
    robot='xmate3_robotiq',
    shader_dir="rt",render_mode="rgb_array")      

import numpy as np
import mediapy
from transforms3d.euler import quat2euler

seed = 0
obs, _ = env.reset(seed=seed)

image = env.render()

if 1:
    radian = -np.pi/2
    nums = 30
    obs = env.get_obs()
    q = obs['extra']['tcp_pose'][3:]
    start_pose = obs['extra']['tcp_pose']    
    action = [0, 0, 0, radian/nums, 0, 0, -1] # rx
    # action = [0, 0, 0, 0, radian/nums, 0, -1] # ry
    # action = [0, 0, 0, 0,  0, radian/nums, -1] # rz
    action = np.asarray(action)
    total_action  = None
    framers = list()
    for i in range(nums):
        # print(action)
        obs, reward, terminated, truncated, info = env.step(action)
        if total_action is None:
            total_action = action.copy()
        else:
            total_action += action.copy()
        framers.append(env.render())
  
    image = env.render()
    q = obs['extra']['tcp_pose'][3:]
    end_pose = obs['extra']['tcp_pose']
    end_euler = np.array(quat2euler(np.array(end_pose[3:])) )
    start_euler = np.array(quat2euler(np.array(start_pose[3:])))
    print(start_euler)
    print(end_euler)
    mediapy.write_video('video_rotate_rx.mp4',framers,fps=30)
    print('ee euler bias',  np.array(end_euler) - np.array(start_euler))

env.close()
del env
`

swearos, 2024-01-29:
@StoneT2000 I found that it is **still wrong** to change the node to xx, but compared with the default panda result, the origin of rx rotation is wrong. Compared with the gripper node in panda_v2.urdf,

https://github.com/haosulab/ManiSkill2/assets/10892059/bfe539e3-472a-4f27-b0fa-605f93f95dce

https://github.com/haosulab/ManiSkill2/blob/d04c2dc5bfd826172b2f3193cbf232a4a33bdd9e/mani_skill2/assets/descriptions/pan da_v2_gripper.urdf#L25C15-L25C29
`
  
  <link name="panda_hand_tcp"/>
  <joint name="panda_hand_tcp_joint" type="fixed">
    <origin rpy="0 0 0" xyz="0 0 0.1034"/>
    <parent link="panda_hand"/>
    <child link="panda_hand_tcp"/>
  </joint>

`

Whether to **add child nodes to robotiq_arg2f_base_link** and then add Z-axis bias

StoneT2000, 2024-02-21:
sorry for the delay. You are correct you should use the convenient link, which basically is the location of the TCP for easier EE control. It seems like the xmate3 robot actually has a small bug/typo in its modelling where it changed the rotation frame. Panda gripper as you point out uses rpy="0 0 0" for the panda_tcp joint, but it is not 0 0 0 for xmate3. For a fix on your side you can edit the URDF file for the xmate3 like so

Original
```
  <link name="grasp_convenient_link"/>
  <joint name="grasp_convenient_joint" type="fixed">
    <origin rpy="0 -1.57079 0" xyz="0 0 0.225"/>
    <parent link="robotiq_arg2f_base_link"/>
    <child link="grasp_convenient_link"/>
  </joint>
  ```
  
  Changed
  ```
  <link name="grasp_convenient_link"/>
  <joint name="grasp_convenient_joint" type="fixed">
    <origin rpy="0 0 0" xyz="0 0 0.225"/>
    <parent link="robotiq_arg2f_base_link"/>
    <child link="grasp_convenient_link"/>
  </joint>
```

You can probably find this file in `data/robots/xmate3_robotiq/xmate3_robotiq.urdf` where `data` is the path to your ms2 assets. data is the default location.

Because other people have potentially trained policies already with the action dimensions permuted wrong for xmate3, i will not update the URDF that is downloaded by default for now. I will update this for the next version of ManiSkill.

closed 2024-06-05 by StoneT2000