mani-skill/ManiSkill #200: An error occurred when I used ray tracing to render on the A100, [svulkan2] [error] Failed to denoise OptiX Error: OPTIX_ERROR_INVALID_VALUE

swearos, 2024-01-31:
An error occurred when I used ray tracing to render on the A100 .
The error is as follows
```
[svulkan2] [error]  OptiX Error: OPTIX_ERROR_INVALID_VALUE
[svulkan2] [error] Failed to denoise OptiX Error: OPTIX_ERROR_INVALID_VALUE
```

1. Vulkan configuration see https://haosulab.github.io/ManiSkill2/getting_started/installation.html#vulkan, 
2. gpu driver has been upgraded to 535.104.05,

Can you share your environment configuration on the A100, ubuntu, cuda version,gpu driver, etc., 
or do you have any other suggestions，thanks!

StoneT2000, 2024-02-02:
@fbxiang any idea?

fbxiang, 2024-02-02:
OptiX is an NVIDIA framework directly built into the driver. These errors sometimes pop up randomly and go away after a restart. In general, something is not quite right with your NVIDIA driver/GPU setup. Unfortunately I do not have a good way to debug such issue.

swearos, 2024-02-04:
I found the problem and fixed it.
when I created docker, I did not mount **/usr/share/nvidia**, so I could not find **nvoptix.bin**. 
[Refer to ](https://github.com/mitsuba-renderer/mitsuba3/issues/1019#issuecomment-1864827711)

closed 2024-02-04 by swearos

kingchou007, 2024-12-25:
hi @swearos 

I have the same problem. Could you give me more information how you solve it?

 `python -m mani_skill.examples.demo_random_action -e PickCube-v1 --render-mode="human" --shader="rt-fast"`
 
 info {'elapsed_steps': tensor([67], dtype=torch.int32), 'success': tensor([False]), 'is_obj_placed': tensor([False]), 'is_robot_static': tensor([False]), 'is_grasped': tensor([False])}
[2024-12-25 22:47:18.104] [svulkan2] [error] Failed to denoise OptiX Error: OPTIX_ERROR_INVALID_VALUE

swearos, 2024-12-26:
@kingchou007 YYou can refer to my last reply, with links to other people's discussions. 
Add -e NVIDIA_DRIVER_CAPABILITIES=all when creating a docker image

SajjadPSavoji, 2025-03-11:
On top the suggestions in this thread, I also had to reinstall/upgrade my nvidia driver and my cuda toolkit.
So just keep in mind if none of the solutions worked for you, it's always worth trying a clean start LOL
Here is a decent tutorial to guide you for that. 
[install nvidia drivers and cuda toolkit for ubuntu](https://www.cherryservers.com/blog/install-cuda-ubuntu)

StoneT2000, 2025-03-11:
^ often times people have incorrectly installed drivers. It appears to be the most common issue and solution. The way the blog installs drivers (apt install) is also our recommended way to do clean installs.

another recommendation is if its driver related we once resolved the issue by downgrading nvidia driver version as some versions do not play nice with vulkan (see https://github.com/KhronosGroup/Vulkan-Tools/issues/1026)