mani-skill/ManiSkill #212: Creating config for xarm6 robot

lauweiquan, 2024-02-14:
Hi,

I have created an environment similar to the excavate demonstration, and I have used the trajectory from excavate to use in replay_trajectory.py and have tried it with no control mode and also pd_ee_target_delta_pose. However, after changing the robot to xarm6, the robot did not move as intended by the trajectory. I believe I may have configured xarm6 wrongly. Do you have some guide on how to configure xarm6 or other robotic arm? 

Thank you.

https://github.com/haosulab/ManiSkill2/assets/1906289/c0a11f8e-4e40-446b-b049-f0fa3ff1d9ab
[https://drive.google.com/file/d/1ZQRuq_DvCHf6oJlUkvoJFNIQ_cTttlcK/view?usp=sharing](url)

StoneT2000, 2024-02-21:
Do you have the URDF file of the xarm6? What are the joints that you can control?

lauweiquan, 2024-02-21:
This is the urdf: https://github.com/lauweiquan/ManiSkill2/blob/main/mani_skill2/assets/descriptions/xarm_description/xarm6_with_gripper_spoon.urdf

using demo_manual_control, i can control all the joints except for the gripper.

StoneT2000, 2024-03-11:
Sorry for the late reply, our team is currently held up with other work at the moment. Can you send the agent class you made for the robot?

lauweiquan, 2024-03-13:
Hi, this is the config: https://github.com/lauweiquan/ManiSkill2/blob/main/mani_skill2/agents/configs/xarm6/defaults.py

and this is the agent file for the robot: https://github.com/lauweiquan/ManiSkill2/blob/main/mani_skill2/agents/robots/xarm6.py

StoneT2000, 2025-02-04:
Tracking the issue over on #635

closed 2025-02-04 by StoneT2000