mani-skill/ManiSkill #223: Vectorized Environment Support for Soft-Body Tasks

TongZhangTHU, 2024-03-05:
Hi,

Thanks for your wonderful work!

I've observed that soft-body simulations are significantly slower compared to rigid-body ones during environment rollouts. While ManiSkill2 offers vectorized environments for rigid-body tasks, it seems there's no support for soft-body scenarios. Can you recommend any strategies to improve the efficiency of soft-body rollouts? Are there any implementations of a vectorized environment for soft-body tasks, possibly utilizing either gym or the ManiSkill2 framework?

StoneT2000, 2024-03-13:
Soft body simulation will inherently be slower and use more GPU resources than a typical rigid body task. Unfortunately for the current maniskill version 2 I'm not aware of any vectorization tool for soft body sim yet that works well.

We are improving it to be much faster in version 3 of maniskill as we develop a new soft body simulator.

TongZhangTHU, 2024-03-13:
@StoneT2000 Thank you for your reply!

closed 2024-03-13 by TongZhangTHU