mani-skill/ManiSkill #246: MPC in ManiSkill2

Altriaex, 2024-03-15:
Hi, I am also interested in the use of MPC in ManiSkill2. I have this question because you mention that some of the demonstrations were collected using MPC.

In this issue (https://github.com/haosulab/ManiSkill2/issues/156) you provide some code for an implementation. But I wonder if it can be used directly with ManiSkill, since it introduces some dependency (eg. from ..builder import MPC).

Could you please provide some scripts/code for generating a demonstration using MPC? I think this will help newbies to understand how the data are collected, and how to use such an algorithm for your tasks.

StoneT2000, 2024-03-28:
@xuanlinli17 can u find the MPC code from before without the missing dependencies?

Also @Altriaex I'm aware ManiSkill2 is missing a bit in terms of example workflows (e.g. demo generation), a example MPC should be added in the next iteration of ManiSkill which we plan to release in the next few months

xuanlinli17, 2024-03-28:
Hi @Altriaex ,

I think you can replace the "from pyrl.xxx import xxx" with "from maniskill2_learn.xxx import xxx". For `from ..builder import MPC`, you can ignore such import and those "@MPC.register_module()` for now, and provide the states and rewards from the state observation mode of ManiSkill2 environments as input to the agent.

Best,

StoneT2000, 2024-05-23:
Hi @Altriaex has this resolved your issue?

closed 2024-05-31 by Altriaex