mani-skill/ManiSkill #257: HTTP Error 403 with Maniskill's VecEnv

RomDeffayet, 2024-04-03:
Hi, 

the VecEnv implementation doesn't seem to work on my machine:

python -m mani_skill2.examples.demo_random_action
 -> works fine
python -m mani_skill2.examples.demo_vec_env -e PickCube-v0 -n 4 --vecenv-type "gym"
 -> works fine
python -m mani_skill2.examples.demo_vec_env -e PickCube-v0 -n 4
 -> following error:

> 024-04-03 11:35:48,301 - mani_skill2 - ERROR - connections to all backends failing; last error: UNKNOWN: ipv4:XX.XX.XX.XX:3128: HTTP proxy returned response code 503
Traceback (most recent call last):
  File "/home/rdeffaye/miniforge3/envs/mani/lib/python3.10/site-packages/mani_skill2/vector/vec_env.py", line 58, in _worker
    env = env_fn()
  File "/home/rdeffaye/miniforge3/envs/mani/lib/python3.10/site-packages/mani_skill2/vector/registration.py", line 16, in _make_env
    env = env_spec.make(**kwargs)
  File "/home/rdeffaye/miniforge3/envs/mani/lib/python3.10/site-packages/mani_skill2/utils/registration.py", line 34, in make
    return self.cls(**_kwargs)
  File "/home/rdeffaye/miniforge3/envs/mani/lib/python3.10/site-packages/mani_skill2/envs/pick_and_place/pick_cube.py", line 22, in __init__
    super().__init__(*args, **kwargs)
  File "/home/rdeffaye/miniforge3/envs/mani/lib/python3.10/site-packages/mani_skill2/envs/pick_and_place/base_env.py", line 27, in __init__
    super().__init__(*args, **kwargs)
  File "/home/rdeffaye/miniforge3/envs/mani/lib/python3.10/site-packages/mani_skill2/envs/sapien_env.py", line 188, in __init__
    obs, _ = self.reset(seed=2022, options=dict(reconfigure=True))
  File "/home/rdeffaye/miniforge3/envs/mani/lib/python3.10/site-packages/mani_skill2/envs/sapien_env.py", line 485, in reset
    self.reconfigure()
  File "/home/rdeffaye/miniforge3/envs/mani/lib/python3.10/site-packages/mani_skill2/envs/sapien_env.py", line 361, in reconfigure
    self._setup_scene()
  File "/home/rdeffaye/miniforge3/envs/mani/lib/python3.10/site-packages/mani_skill2/envs/sapien_env.py", line 621, in _setup_scene
    self._scene = self._engine.create_scene(scene_config)
RuntimeError: connections to all backends failing; last error: UNKNOWN: ipv4:XX.XX.XX.XX:3128: HTTP proxy returned response code 503

Did I miss something ?
Thanks :pray:

StoneT2000, 2024-04-04:
Which version of maniskill are you running and what linux system are you on?

RomDeffayet, 2024-04-05:
Sure, I'm running Maniskill 0.5.3 on RHEL 8.8, Driver Version: 535.104.05   CUDA Version: 12.2

StoneT2000, 2024-04-30:
Apologies for the very late followup. We are transitioning to ManiSkill 3 very soon so we recommend to test on that version once it is out (it will probably fix this potential bug)

StoneT2000, 2024-05-02:
@RomDeffayet we have just released a beta version of ManiSkill 3, which shouldn't have these issues as it doesn't rely on multiprocessing for env vectorization. Let me know if it works for you

RomDeffayet, 2024-05-03:
Hi, I just tried with the beta version and it is indeed solved :raised_hands: 
Thanks

closed 2024-05-03 by RomDeffayet