mani-skill/ManiSkill #268: make new soft body

nutsintheshell, 2024-04-10:
Dear authors, 
Your work is pretty good. I'd like to make some soft bodies in your environment. I'm wondering is there a way to make my own soft models or load a soft model from other paper. Thanks for your reply.

StoneT2000, 2024-05-23:
@fbxiang

fbxiang, 2024-05-23:
The MPM based soft body is currently available in ManiSkill 2 (v0.5.3 branch of this repo). I assuming you are using that branch, you can take a look at one of the soft body envs to see how soft bodies are added. Essentially at initialization, we add particles uniformly distributed in a volume into the scene. For example, the Pour env [samples particles from a cylinder](https://github.com/haosulab/ManiSkill/blob/493be36121a9dd06071a57172274babe617b789f/mani_skill2/envs/mpm/pour_env.py#L178-L191). There is no standard file formats for creating soft body, but you can always just create some procedure to sample from mesh volumes and use regular watertight meshes as your soft body representation.

nutsintheshell, 2024-05-23:
> The MPM based soft body is currently available in ManiSkill 2 (v0.5.3 branch of this repo). Assuming you are using that branch, you can take a look at one of the soft body envs to see how soft bodies are added. Essentially at initialization, we add particles uniformly distributed in a volume into the scene. For example, the Pour env [samples particles from a cylinder](https://github.com/haosulab/ManiSkill/blob/493be36121a9dd06071a57172274babe617b789f/mani_skill2/envs/mpm/pour_env.py#L178-L191). There is no standard file formats for creating soft body, but you can always just create some procedure to sample from mesh volumes and use regular watertight meshes as your soft body representation.

Thanks for your reply. I 've tried to make my soft body successfully.

closed 2025-02-04 by StoneT2000

yang-si-hang, 2025-05-13:
@fbxiang. I was wondering if it's possible to add my own custom soft bodies to the "ManiSkill3" environment programmatically. Specifically, can this be done using the MPM method you mentioned, or is that feature currently only available in "ManiSkill2"?

StoneT2000, 2025-05-13:
The MPM code in maniskill2 is being ported to maniskill3 at the moment. You currently have to use maniskill 2 to do soft body simulation. The ManiSkill3 MPM will be faster and majorly improved compared to ms2, mostly in terms of documentation and API design.

yang-si-hang, 2025-05-13:
Thanks so much! I saw you mentioned that soft body simulation will be added to ManiSkill3 soon. Do you have an idea of when this feature might be released later this year?

StoneT2000, 2025-05-13:
Maybe in the summer. No hard timeline yet sorry

yang-si-hang, 2025-05-14:
Got it, thanks for the update! Looking forward to it.