mani-skill/ManiSkill #278: VK_KHR_xcb_surface extension missing error

YandongJi, 2024-04-20:
Hi,

I'm having this error when running `python -m mani_skill2.examples.demo_random_action`:

**`[2024-04-20 11:36:31.505] [svulkan2] [error] GLFW error: X11: Vulkan instance missing VK_KHR_xcb_surface extension`**

I updated Vulkan SDK, Vulkan Driver, the extension exists as "VK_KHR_xcb_surface : extension revision 6" as well as

"Surface types: count = 2
VK_KHR_xcb_surface
VK_KHR_xlib_surface" on RTX 4090 GPU, I tried installing "sudo apt-get install libxcb1-dev libx11-xcb-dev".

I exported these path in bashrc: 
```
export VK_ICD_FILENAMES=/usr/share/vulkan/icd.d/nvidia_icd.json #:/usr/share/vulkan/icd.d/intel_icd.x86_64.json:/usr/share/vulkan/icd.d/lvp_icd.x86_64.json
export VK_LAYER_PATH=/usr/share/vulkan/explicit_layer.d/:/usr/share/vulkan/icd.d/
export VK_INSTANCE_EXTENSIONS=VK_KHR_surface:VK_KHR_xcb_surface
```
But this error still exists. Looking for kind help now.....

StoneT2000, 2024-04-20:
@fbxiang

fbxiang, 2024-04-20:
Are you saying it works in Docker but not outside of Docker? Or you are using some other visualization method?
Also I think the variable VK_INSTANCE_EXTENSIONS should not be set.
You can follow the following steps to debug:
1. Try to get the program `vkcube` to work (you can probably find more resources on how to debug vkcube). If vkcube won't run on screen,  I cannot really give any useful advise.
2. Test maniskill again, if it still does not work, paste the result of `vulkaninfo` here

StoneT2000, 2024-05-23:
Hi @YandongJi do you still have this issue?

YandongJi, 2024-05-23:
yes, thanks for checking, it still only works in docker, I can look into this a week later

StoneT2000, 2024-08-01:
Closing the issue now as it has become stale.

closed 2024-08-01 by StoneT2000