mani-skill/ManiSkill #312: mani_skill.examples.demo_reset_distribution is not executable

BXZZcj, 2024-05-03:
The script 'python mani_skill.examples.demo_reset_distribution' is not executable.

I ran the command "python -m mani_skill.examples.demo_reset_distribution -e "PegInsertionSide-v1"" following the instructions in the ManiSkill3 documentation (https://maniskill.readthedocs.io/en/latest/user_guide/demos/index.html#gpu-simulation), and encountered the following errors:

```
AttributeError: 'PegInsertionSideEnv' object has no attribute 'render_images' 
```

Even after changing the env_id from "PegInsertionSide-v1" to "StackCube-v1", the error persists.

StoneT2000, 2024-05-03:
Looks like documentation wrong / script could be a bit more robust. 

To fix you have to pass in 

--record-dir=videos

In addition to the environment ID, then it should save a video of samples initial states

BXZZcj, 2024-05-03:
> Looks like documentation wrong / script could be a bit more robust.
> 
> To fix you have to pass in
> 
> --record-dir=videos
> 
> In addition to the environment ID, then it should save a video of sampled initial states

Thank you for your reply. That fixes the error. 

My bad. I didn't carefully read the code before. "--render-mode="sensors"" is needed for the GUI. But perhaps that parameter should be added to the corresponding command as shown in the document. Thank you.

StoneT2000, 2024-05-03:
I think smth we need to fix is to try and get all example scripts to have the same arguments, they are currently all standalone scripts so there are some discrepancies. Thanks again for finding the bug!

PR #313 (2024-05-03): [Docs] Fix docs for visualizing reset distributions demo
Closes #312
closed 2024-05-03 by StoneT2000