mani-skill/ManiSkill #316: Command in the Document not Executable

BXZZcj, 2024-05-05:
The command "python -m test.py -r "my_panda" -c "pd_joint_pos" --keyframe-actions" in the document (https://maniskill.readthedocs.io/en/latest/user_guide/tutorials/custom_robots.html) is not executable.

StoneT2000, 2024-05-05:
There is a typo here, if you follow the tutorial you should have a test.py file. To run get rid of the -m. Should fix it

StoneT2000, 2024-05-06:
This is intentional. Typically grippers are mimic joints, where there is a constraint on the joint angle to be the same. For Panda the implementation we provide uses mimic joint controller so there is one less degree of freedom. If you  don't use the mimic joint controller config and use the normal pd joint position one then you can control both gripper joints

StoneT2000, 2024-05-06:
ok this has been fixed now. The demo robot code (test.py) will now handle situations where a robot might have multiple joints constrained against each other and still accept an action that applies a qpos to every joint. To try it out you can install directly from github or you can wait for the next pypi release.

closed 2024-05-06 by StoneT2000

BXZZcj, 2024-05-07:
> ok this has been fixed now. The demo robot code (test.py) will now handle situations where a robot might have multiple joints constrained against each other and still accept an action that applies a [...]

Thank you for your attention.

PR #319 (2024-05-06): fix bug for demo robot code
closes #316