mani-skill/ManiSkill #324: Error When Running the test Installation code

eyucherin, 2024-05-07:
Hello I am trying to run the test installation but I get the following error. Any insight on this would be greatly appreciated! 

```
(base) cherin@cherin-MS-7D99:~$ python -m mani_skill.examples.demo_random_action
Traceback (most recent call last):
  File "/usr/anaconda3/lib/python3.8/runpy.py", line 194, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/usr/anaconda3/lib/python3.8/runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "/usr/anaconda3/lib/python3.8/site-packages/mani_skill/examples/demo_random_action.py", line 6, in <module>
    from mani_skill.envs.sapien_env import BaseEnv
  File "/usr/anaconda3/lib/python3.8/site-packages/mani_skill/envs/__init__.py", line 1, in <module>
    from .scenes import *
  File "/usr/anaconda3/lib/python3.8/site-packages/mani_skill/envs/scenes/__init__.py", line 1, in <module>
    from mani_skill.utils.registration import register_env
  File "/usr/anaconda3/lib/python3.8/site-packages/mani_skill/utils/registration.py", line 12, in <module>
    from mani_skill.vector.wrappers.gymnasium import ManiSkillVectorEnv
  File "/usr/anaconda3/lib/python3.8/site-packages/mani_skill/vector/wrappers/gymnasium.py", line 6, in <module>
    import torch
  File "/usr/anaconda3/lib/python3.8/site-packages/torch/__init__.py", line 237, in <module>
    from torch._C import *  # noqa: F403
ImportError: libnccl.so.2: cannot open shared object file: No such file or directory

```

StoneT2000, 2024-05-07:
Can you run pip show sapien?

this will most likely be solved by using Python >=3.9, I believe the sapien package does not support 3.8

eyucherin, 2024-05-08:
Hi!
It seems like I am running it on python 3.8.  

```
(base) cherin@cherin-MS-7D99:~$ pip show sapien
Name: sapien
Version: 2.2.2
Summary: ['SAPIEN: A SimulAted Parted based Interactive ENvironment']
Home-page: https://sapien.ucsd.edu
Author: Sapien
Author-email: sapienaicontact@gmail.com
License: MIT
Location: /usr/anaconda3/lib/python3.8/site-packages
Requires: transforms3d, requests, opencv-python, numpy
Required-by: mani-skill2, mani-skill
```

closed 2024-05-08 by eyucherin