mani-skill/ManiSkill #342: Only 2 tasks available when downloading demonstrations

YY-GX, 2024-05-22:
Hi all, when I use `python -m mani_skill.utils.download_demo` to check what packages I could download, there're only 2:
```shell
Available dataset UIDs:
['PickCube-v1', 'StackCube-v1']
``` 
The size of them are very small as well:
```shell
480K	PickCube-v1
600K	StackCube-v1
```
Could you please help me with this? It's really appreciated!!

StoneT2000, 2024-05-22:
I can upload more demonstrations later. At the moment the two you downloaded are the human teleoperated datasets (hence small size)

Last time we had motion planning generatable datasets which can be generated following this documentation: https://maniskill.readthedocs.io/en/latest/user_guide/data_collection/motionplanning.html

I will upload a downloadable result of the demo generation scripts in the future as well as RL generated demos for some tasks

StoneT2000, 2024-05-22:
One reason I have not yet uploaded all demo datasets is because we are still discussing how to best parametrize environment state when  objects in the environment may have different geometries and are seed dependent.

StoneT2000, 2024-06-09:
Hi @YY-GX I have uploaded more demonstrations, some created via different sources (motion planning, teleop guided motion planning by human, RL/neural network generated). There are now 5 tasks available for download (PushCube, PickCube, StackCube, PlugCharger, PegInsertionSide) if you use the main code branch. Alternatively you can download them straight from hugging face https://huggingface.co/collections/haosulab/maniskill-demonstration-datasets-66565c3c37a090c729dbdb16

closed 2024-06-09 by StoneT2000