mani-skill/ManiSkill #344: Expose Joint Force/Torque values

StoneT2000, 2024-05-23:
Request from here https://github.com/haosulab/SAPIEN/discussions/156

Currently it is in the nightly package of SAPIEN but not completely tested yet with ManiSkill / proper API developed for it.

jstmn, 2025-05-05:
Hey @StoneT2000, 

Where can I find this functionality? I'm not sure where the nightly package is!

Thanks

jstmn, 2025-05-05:
Hey @StoneT2000 ,

Are there any plans to get `link_incoming_joint_forces()` and `get_link_incoming_joint_forces()` from here https://github.com/haosulab/SAPIEN/commit/3037022b43a7466d037dd61877c72e4b18f8aacc added to the main ManiSkill release?

StoneT2000, 2025-05-07:
Yes we can expose those APIs

closed 2025-05-12 by StoneT2000

StoneT2000, 2025-05-13:
@jstmn lmk if the api works as expected. No one on our team uses it so it's difficult to make any reasonable tests for it

PR #1061 (2025-05-12): [Feature] Expose API for incoming link joint forces on CPU and GPU sim
Closes #344