mani-skill/ManiSkill #346: Fix replay tool for converting pd_joint_pos to pd_ee_delta_pos/pose control

StoneT2000, 2024-05-28:


sean1295, 2024-06-19:
Hello, thank you so much for your contribution to making this amazing environment available! I am curious if you have had any time to fix the replay trajectory code. I am trying to do imitation learning with motion planned trajectories but it looks like transforming it into eef control is impossible. Thanks in advance.

StoneT2000, 2024-06-21:
yeah this is on my TODO list at the moment. Will bump up the priority to try and fix it earlier

StoneT2000, 2024-06-22:
@sean1295 should be mostly fixed now. For less precise tasks like PickCube, the success rate of conversion to pd_ee_delta_pos/pose is near 100%. For highly precise tasks like PegInsertion and PlugCharger it is lower.

PR #388 (2024-06-22): [BugFix] Fix PD joint position to PD ee delta pose control conversion
Fixes #346
closed 2024-06-22 by StoneT2000