mani-skill/ManiSkill #351: Question about benchmarks

btaba, 2024-05-30:
Maniskill is an awesome repo, really appreciate the work!

QQ: When running the GPU benchmarks, if I increase the sim-freq but keep the same ratio of sim to control freq, the steps-per-second increases significantly.

For example:

```
python -m mani_skill.examples.benchmarking.gpu_sim -e "PickCube-v1" -n=4096 -o=state --control-freq=20 --sim-freq=100
# -------------------------------------------------------------------------- #
Task ID: PickCube-v1, 4096 parallel environments, sim_backend=gpu
obs_mode=state, control_mode=pd_joint_delta_pos
render_mode=sensors, sensor_details=RGBD(128x128)
sim_freq=100, control_freq=20
....
# -------------------------------------------------------------------------- #
start recording env.step metrics
env.step: 44241.397 steps/s, 10.801 parallel steps/s, 100 steps in 9.258s

# ---

python -m mani_skill.examples.benchmarking.gpu_sim -e "PickCube-v1" -n=4096 -o=state --control-freq=125 --sim-freq=500
# -------------------------------------------------------------------------- #
Task ID: PickCube-v1, 4096 parallel environments, sim_backend=gpu
obs_mode=state, control_mode=pd_joint_delta_pos
render_mode=sensors, sensor_details=RGBD(128x128)
sim_freq=500, control_freq=125
...
# -------------------------------------------------------------------------- #
start recording env.step metrics
env.step: 85680.663 steps/s, 20.918 parallel steps/s, 100 steps in 4.781s

```

So for control-freq/sim-freq 20/100, I get ~44k steps/s, but  for 125/500, I get ~86k steps/s.  I would expect these two numbers to be roughly the same, is there any insight for why higher sim-freq results in higher throughput? Maybe something with contact/constraint solver caching?

StoneT2000, 2024-05-30:
Thank you for the kind comment!

So in ManiSkill, there are two temporal levels of simulation. There is the underlying higher frequency physics simulation powered by physx, and then there is the lower frequency control simulation where each step of control is one applied action (target positions/velocities are applied to active joints).

In ManiSkill, we take one control step and apply the given actions and then take sim_freq / control_freq physics steps.

High sim frequency affects the `dt` configuration, saying how much time passes when taking a physics step. If sim freq is higher, then less time is taken between physics steps and I imagine this means less solving is needed.

As a result you can observe the following if you add --save-video do your two commands (and lower number of envs to something storeable in a  video)

Top video is sim, control freq of 500, 100, bottom video is 100, 20 (ratios are kept the same)

https://github.com/haosulab/ManiSkill/assets/35373228/d92f706b-1941-437b-a862-39d8fc553894

https://github.com/haosulab/ManiSkill/assets/35373228/47172829-72da-4604-aaa4-06359f781b13

Each frame of the video is a single control step, comprised of 100 / 20 = 500 / 100 = constant 5 physics steps. Hence with higher sim freq it looks smoother/slower per control step since less time is passing, lower sim freq more time passes between control steps (and solver is less accurate)

btaba, 2024-05-30:
Awesome, thanks for the insight!

So it sounds like as the physics step time (`dt`) is increased,  PhysX takes a longer amount of wall time to do a single physics step (potentially due to the constraint solver iterations being dynamic and dependent on `dt`?).  That's quite interesting, any further insight on the PhysX side would be appreciated.

Closing the issue for now, thanks again!

closed 2024-05-30 by btaba

StoneT2000, 2024-05-31:
PhysX internals are a bit out of my scope of understanding, @fbxiang might understand it better but tbh PhysX is not super well documented.

https://nvidia-omniverse.github.io/PhysX/physx/5.3.1/docs/Simulation.html is the docs about the whole substepping thing, could not find more details on `dt`.

fbxiang, 2024-05-31:
PhysX is known to do less solver iterations with smaller dt. And smaller time steps in general makes a lot of simulation problems easier to solve.