mani-skill/ManiSkill #357: [Question/Bug] Missing file (and folder) `mani_skill2_ycb/info_pick_v0.json'

CreativeNick, 2024-06-01:
I attempted running the command `python -m mani_skill.examples.demo_random_action -e "PickSingleYCB-v1" --render-mode="human"`, however, I received the following error:

### Command I ran:
`python -m mani_skill.examples.demo_random_action -e "PickSingleYCB-v1" --render-mode="human"`

### Output:
```
(ms_dev) creativenick@creativenick:~/Desktop$ python -m mani_skill.examples.demo_random_action -e "PickSingleYCB-v1" --render-mode="human"
opts: []
env_kwargs: {}
Traceback (most recent call last):
  File "/home/creativenick/anaconda3/envs/ms_dev/lib/python3.9/runpy.py", line 197, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/home/creativenick/anaconda3/envs/ms_dev/lib/python3.9/runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "/home/creativenick/Desktop/ManiSkill/mani_skill/examples/demo_random_action.py", line 98, in <module>
    main(parse_args())
  File "/home/creativenick/Desktop/ManiSkill/mani_skill/examples/demo_random_action.py", line 47, in main
    env: BaseEnv = gym.make(
  File "/home/creativenick/anaconda3/envs/ms_dev/lib/python3.9/site-packages/gymnasium/envs/registration.py", line 802, in make
    env = env_creator(**env_spec_kwargs)
  File "/home/creativenick/Desktop/ManiSkill/mani_skill/utils/registration.py", line 82, in make
    env = env_spec.make(**kwargs)
  File "/home/creativenick/Desktop/ManiSkill/mani_skill/utils/registration.py", line 35, in make
    return self.cls(**_kwargs)
  File "/home/creativenick/Desktop/ManiSkill/mani_skill/envs/tasks/tabletop/pick_single_ycb.py", line 47, in __init__
    load_json(ASSET_DIR / "assets/mani_skill2_ycb/info_pick_v0.json").keys()
  File "/home/creativenick/Desktop/ManiSkill/mani_skill/utils/io_utils.py", line 31, in load_json
    f = open(filename, "rt")
FileNotFoundError: [Errno 2] No such file or directory: '/home/creativenick/.maniskill/data/assets/mani_skill2_ycb/info_pick_v0.json'
```

I tried looking in the repo for `info_pick_v0.json` but was unsuccessful. I also noticed that it was looking for the file inside a folder called `/mani_skill2_ycb`. Could the reason for this error be that `info_pick_v0.json` is from the older version of ManiSkill (maniskill2) and was not updated to ManiSkill3, which is why the folder `/mani_skill2_ycb` and the file `info_pick_v0.json` are both missing? Or perhaps I missed something else entirely?

Any help/insights would be appreciated, thanks!

StoneT2000, 2024-06-01:
Could you try 

python -m mani_skill.utils.download_asset "PickSingleYCB-v1"

It's on the TODO list to auto print message about which datasets are needed for which tasks. Generally running the above command and changing the env id to the one you want will download the necessary assets

CreativeNick, 2024-06-02:
It worked, thanks!

hesic73, 2024-07-25:
I think this issue should be mentioned in the documentation here as well: 
https://maniskill.readthedocs.io/en/latest/user_guide/tutorials/custom_tasks/loading_objects.html#loading-from-existing-datasets

Including a note about the utility to manually download datasets (e.g., `python -m mani_skill.utils.download_asset <DATASET_NAME>`) would help users avoid missing file errors.

StoneT2000, 2024-07-25:
Oops this issue has been sitting there for a while: https://github.com/haosulab/ManiSkill/issues/358

Will get on to this tomorrow probably.

StoneT2000, 2024-07-27:
Ok I can one up the feature. Now when you create a ManiSkill environment, we check for assets. if assets are detected as missing (robot, environment related assets like YCB dataset), then you are prompted to download them and if you say yes they are downloaded and code proceeds as normal (and future runs will not require downloads).

StoneT2000, 2024-07-27:
Ok I can one up the feature. Now when you create a ManiSkill environment, we check for assets. if assets are detected as missing (robot, environment related assets like YCB dataset), then you are prompted to download them and if you say yes they are downloaded and code proceeds as normal (and future runs will not require downloads). 

Run pip install --upgrade git+https://github.com/haosulab/ManiSkill.git to get the latest commit/release to use this. Pypi package is not yet updated with this feature.

PR #456 (2024-07-27): [Feature] Automatic asset download checks
Now all ManiSkill tasks (this does not include user built tasks outside of this repo) will support automatic asset download checks. This is done at env/robot registration time, where you include the data source IDs required to run the env/robots. 

Closes #357
closed 2024-07-27 by StoneT2000