mani-skill/ManiSkill #382: Getting error when reseting environment without `reconfigure=False`

rafaelBauer, 2024-06-14:
**Library version:** mani_skill==3.0.0b4
**Python:** 3.11.0

I am getting the following error when calling env.reset():

```
  File "/home/bauerr/git/project/src/controller/controller.py", line 193, in reset
    self._previous_observation = self.__reset_function()
                                 ^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/bauerr/git/project/src/envs/maniskill.py", line 89, in reset
    observation, _ = self.__env.reset()
                         ^^^^^^^^^^^^^^^^^^
  File "/home/bauerr/miniconda3/envs/project/lib/python3.11/site-packages/gymnasium/wrappers/time_limit.py", line 75, in reset
    return self.env.reset(**kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/bauerr/miniconda3/envs/project/lib/python3.11/site-packages/gymnasium/wrappers/order_enforcing.py", line 61, in reset
    return self.env.reset(**kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/bauerr/miniconda3/envs/project/lib/python3.11/site-packages/mani_skill/envs/sapien_env.py", line 701, in reset
    self.agent.reset()
  File "/home/bauerr/miniconda3/envs/project/lib/python3.11/site-packages/mani_skill/agents/base_agent.py", line 312, in reset
    self.controller.reset()
  File "/home/bauerr/miniconda3/envs/project/lib/python3.11/site-packages/mani_skill/agents/controllers/base_controller.py", line 236, in reset
    controller.reset()
  File "/home/bauerr/miniconda3/envs/project/lib/python3.11/site-packages/mani_skill/agents/controllers/pd_joint_pos.py", line 64, in reset
    self._target_qpos[self.scene._reset_mask] = self.qpos[
    ~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^
RuntimeError: Index put requires the source and destination dtypes match, got Double for the destination and Float for the source.
```

I have the `pd_joint_pos` controller.

Any pointers?
I will check if while the reset() is called I am still sending step commands. Could this be a reason?

rafaelBauer, 2024-06-14:
Digging further, I saw that the Tensor `self.qpos` is a `torch.float32 and the other, `self._target_qpos`, is `torch.float64`.
I made sure that the action I give in the step method is a float32 and it worked

closed 2024-06-14 by rafaelBauer

StoneT2000, 2024-06-15:
Ah thanks for raising this issue. This sounds more like a bug that we should fool-proof so someone sending a float64 is warned about it.

rafaelBauer, 2024-06-15:
Yeah, I think so too.. I have closed the issue myself,  because I didn't want to bother you with it, but maybe you should keep it open.