mani-skill/ManiSkill #387: Unable to train 'PickclutterYCB' using PPO

jayaramreddy10, 2024-06-22:
Hi, 

I am facing an issue when training PPO policy on 'PickClutterYCB' task. 

Command used: python examples/baselines/ppo/ppo.py --env_id="PickClutterYCB-v1" --exp-name="state-pickclutterYCB" --num_envs=1024 --update_epochs=8 --num_minibatches=32 --total_timesteps=5_000_000 --eval_freq=8 --num-steps=20

Issue: 
Traceback (most recent call last):
  File "/home/jai/ManiSkill/examples/baselines/ppo/ppo.py", line 270, in <module>
    returns = np.concatenate(returns)
ValueError: need at least one array to concatenate

Am I using the wrong command line args for training this skill?

StoneT2000, 2024-06-22:
This occurs if the number of evaluation steps is too small and so none of the episodes complete during evaluation.

Try setting num_eval_steps=100 (it has to be at least as high as the max episode steps value of the environment).

jayaramreddy10, 2024-06-22:
Got it, thanks a lot. Its working now.

closed 2024-06-22 by StoneT2000