mani-skill/ManiSkill #395: Using own scenes

clemense, 2024-06-25:
What's the interface in Maniskill to use my own scenes?
The docs https://maniskill.readthedocs.io/en/latest/user_guide/tutorials/custom_reusable_scenes.html are unfortunately still WIP.

StoneT2000, 2024-06-25:
Hi @clemense thanks for raising the issue. Despite that being WIP, you can use your own scenes but you have to write code to read them and build objects inside them. https://github.com/haosulab/ManiSkill/blob/main/mani_skill/utils/scene_builder/replicacad/scene_builder.py is some example code you can try and follow but it requires you to have some understanding of how to build custom actors/articulations (see [these docs](https://maniskill.readthedocs.io/en/latest/user_guide/tutorials/custom_tasks/loading_objects.html))

Currently we have not decided on a "standard" format for scene definitions as multiple past frameworks (Meta's Habitat or AI2's THOR) have different JSON schemas and definitions and I am not an expert on this, so I have left it as WIP for now. We simply provide example code of how to build scenes from AI2 THOR and Habitat.

StoneT2000, 2024-07-18:
Hi @clemense has your issue been resolved?

clemense, 2024-07-18:
Yes, thanks for the pointers! This is good for writing own parsers but I'm looking forward to see some standard format being adopted.

closed 2024-07-18 by clemense