mani-skill/ManiSkill #401: Installation Error

HeegerGao, 2024-07-01:
Hi, I am installing ManiSkill on my headless server. After installing mani_skill and torch with pip, I run `python -m mani_skill.examples.demo_random_action` and then I get the following error:

```
opts: []
env_kwargs: {}
Traceback (most recent call last):
  File "/home/gck/.conda/envs/contact/lib/python3.9/runpy.py", line 197, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/home/gck/.conda/envs/contact/lib/python3.9/runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "/home/gck/.conda/envs/contact/lib/python3.9/site-packages/mani_skill/examples/demo_random_action.py", line 106, in <module>
    main(parse_args())
  File "/home/gck/.conda/envs/contact/lib/python3.9/site-packages/mani_skill/examples/demo_random_action.py", line 54, in main
    env: BaseEnv = gym.make(
  File "/home/gck/.conda/envs/contact/lib/python3.9/site-packages/gymnasium/envs/registration.py", line 802, in make
    env = env_creator(**env_spec_kwargs)
  File "/home/gck/.conda/envs/contact/lib/python3.9/site-packages/mani_skill/utils/registration.py", line 82, in make
    env = env_spec.make(**kwargs)
  File "/home/gck/.conda/envs/contact/lib/python3.9/site-packages/mani_skill/utils/registration.py", line 35, in make
    return self.cls(**_kwargs)
  File "/home/gck/.conda/envs/contact/lib/python3.9/site-packages/mani_skill/envs/tasks/tabletop/push_cube.py", line 67, in __init__
    super().__init__(*args, robot_uids=robot_uids, **kwargs)
  File "/home/gck/.conda/envs/contact/lib/python3.9/site-packages/mani_skill/envs/sapien_env.py", line 274, in __init__
    obs, _ = self.reset(seed=2022, options=dict(reconfigure=True))
  File "/home/gck/.conda/envs/contact/lib/python3.9/site-packages/mani_skill/envs/sapien_env.py", line 684, in reset
    self._reconfigure(options)
  File "/home/gck/.conda/envs/contact/lib/python3.9/site-packages/mani_skill/envs/sapien_env.py", line 542, in _reconfigure
    self._setup_scene()
  File "/home/gck/.conda/envs/contact/lib/python3.9/site-packages/mani_skill/envs/sapien_env.py", line 927, in _setup_scene
    sapien.Scene([self.physx_system, sapien.render.RenderSystem()])
RuntimeError: failed to find a rendering device
```

I am using Ubuntu 20.04, with 2 3090 GPUs. CUDA version is 11.7. I am using python 3.9. I am sure the three vulkan json files exist.

StoneT2000, 2024-07-01:
What driver version do you have (check `nvidia-smi`)?

HeegerGao, 2024-07-02:
Thanks for your prompt reply. It's 535.104.05.

```
Tue Jul  2 02:31:35 2024
+---------------------------------------------------------------------------------------+
| NVIDIA-SMI 535.104.05             Driver Version: 535.104.05   CUDA Version: 12.2     |
|-----------------------------------------+----------------------+----------------------+
| GPU  Name                 Persistence-M | Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp   Perf          Pwr:Usage/Cap |         Memory-Usage | GPU-Util  Compute M. |
|                                         |                      |               MIG M. |
|=========================================+======================+======================|
|   0  NVIDIA GeForce RTX 3090        On  | 00000000:03:00.0 Off |                  N/A |
|  0%   41C    P8              19W / 370W |      3MiB / 24576MiB |      0%      Default |
|                                         |                      |                  N/A |
+-----------------------------------------+----------------------+----------------------+
|   1  NVIDIA GeForce RTX 3090        On  | 00000000:04:00.0  On |                  N/A |
|  0%   35C    P8              19W / 370W |      3MiB / 24576MiB |      0%      Default |
|                                         |                      |                  N/A |
+-----------------------------------------+----------------------+----------------------+

+---------------------------------------------------------------------------------------+
| Processes:                                                                            |
|  GPU   GI   CI        PID   Type   Process name                            GPU Memory |
|        ID   ID                                                             Usage      |
|=======================================================================================|
|  No running processes found                                                           |
+---------------------------------------------------------------------------------------+
```

StoneT2000, 2024-07-02:
@fbxiang any idea?

fbxiang, 2024-07-02:
If the machine has a Window display, maybe debugging with `vkcube` could help (try and see if `vkcube` can run under the NVIDIA GPU). If it is on a server. It would be a bit harder to debug. Maybe I can take a look at the output of `sapien info --all` and `vulkaninfo`.

HeegerGao, 2024-07-03:
Thank @fbxiang . Here are the outputs of `sapien info --all` and `vulkaninfo`. It seems like there are indeed some problems.

```
~$ sapien info --all
Traceback (most recent call last):
  File "/home/gck/.conda/envs/contact/bin/sapien", line 58, in <module>
    info += sapien.render.get_device_summary() + "\n"
RuntimeError: failed to find a rendering device
```

```
~$ vulkaninfo
ERROR: [Loader Message] Code 0 : libGLX_nvidia.so.0: cannot open shared object file: No such file or directory
'DISPLAY' environment variable not set... skipping surface info
error: XDG_RUNTIME_DIR not set in the environment.
error: XDG_RUNTIME_DIR not set in the environment.
error: XDG_RUNTIME_DIR not set in the environment.
error: XDG_RUNTIME_DIR not set in the environment.
error: XDG_RUNTIME_DIR not set in the environment.
WARNING: lavapipe is not a conformant vulkan implementation, testing use only.
==========
VULKANINFO
==========

Vulkan Instance Version: 1.2.131

Instance Extensions: count = 18
====================
        VK_EXT_acquire_xlib_display            : extension revision 1
        VK_EXT_debug_report                    : extension revision 10
        VK_EXT_debug_utils                     : extension revision 1
        VK_EXT_direct_mode_display             : extension revision 1
        VK_EXT_display_surface_counter         : extension revision 1
        VK_KHR_device_group_creation           : extension revision 1
        VK_KHR_display                         : extension revision 23
        VK_KHR_external_fence_capabilities     : extension revision 1
        VK_KHR_external_memory_capabilities    : extension revision 1
        VK_KHR_external_semaphore_capabilities : extension revision 1
        VK_KHR_get_display_properties2         : extension revision 1
        VK_KHR_get_physical_device_properties2 : extension revision 2
        VK_KHR_get_surface_capabilities2       : extension revision 1
        VK_KHR_surface                         : extension revision 25
        VK_KHR_surface_protected_capabilities  : extension revision 1
        VK_KHR_wayland_surface                 : extension revision 6
        VK_KHR_xcb_surface                     : extension revision 6
        VK_KHR_xlib_surface                    : extension revision 6

Layers: count = 4
=======
VK_LAYER_LUNARG_standard_validation (LunarG Standard Validation Layer) Vulkan version 1.0.131, layer version 1:
        Layer Extensions: count = 0
        Devices: count = 1
                GPU id  : 0 (llvmpipe (LLVM 12.0.0, 256 bits))
                Layer-Device Extensions: count = 0

VK_LAYER_MESA_device_select (Linux device selection layer) Vulkan version 1.2.73, layer version 1:
        Layer Extensions: count = 0
        Devices: count = 1
                GPU id  : 0 (llvmpipe (LLVM 12.0.0, 256 bits))
                Layer-Device Extensions: count = 0

VK_LAYER_MESA_overlay (Mesa Overlay layer) Vulkan version 1.1.73, layer version 1:
        Layer Extensions: count = 0
        Devices: count = 1
                GPU id  : 0 (llvmpipe (LLVM 12.0.0, 256 bits))
                Layer-Device Extensions: count = 0

VK_LAYER_NV_optimus (NVIDIA Optimus layer) Vulkan version 1.2.155, layer version 1:
        Layer Extensions: count = 0
        Devices: count = 1
[... 479 lines cut ...]
        shaderSharedInt64Atomics = true

VkPhysicalDeviceShaderDrawParametersFeatures:
---------------------------------------------
        shaderDrawParameters = true

VkPhysicalDeviceTransformFeedbackFeaturesEXT:
---------------------------------------------
        transformFeedback = true
        geometryStreams   = true

VkPhysicalDeviceUniformBufferStandardLayoutFeaturesKHR:
-------------------------------------------------------
        uniformBufferStandardLayout = true

VkPhysicalDeviceVariablePointersFeatures:
-----------------------------------------
        variablePointersStorageBuffer = true
        variablePointers              = false

VkPhysicalDeviceVertexAttributeDivisorFeaturesEXT:
--------------------------------------------------
        vertexAttributeInstanceRateDivisor     = true
        vertexAttributeInstanceRateZeroDivisor = false
```

fbxiang, 2024-07-03:
libGLX_nvidia.so.0 is NVIDIA's graphics driver and it is required for any kind of rendering. Vulkan info shows that your NVIDIA GPU is not capable of rendering because you are missing this driver. It can probably be fixed by re-install NVIDIA driver (if you use apt, make sure you install nvidia-driver-xxx without the word **server** in its name).

StoneT2000, 2024-07-15:
Checking in @HeegerGao has your problem been resolved?

HeegerGao, 2024-07-16:
Not yet. I am using a commercial cloud server, and it is very hard to reinstall the driver on it. I am trying on that. Actually I can successfully install ManiSkill on my real server.

HeegerGao, 2024-07-19:
Hi, I think this is a hardware-related issue, probably because the hardware virtualization isolation of this specific cloud server is not good. I can successfully install ManiSkill on my other devices. So I do not think this issue has reference value for others and I will close it.

closed 2024-07-19 by HeegerGao

StoneT2000, 2024-07-19:
Yes it is known that some GPU virtualization setups do not work and can't render things at all.