mani-skill/ManiSkill #402: Does maniskill3's renderer supports remote window display

DevinQiao, 2024-07-02:
I install maniskill3 in a remote linux server, and run the example environment ReplicaCAD using the following code:
```
python -m mani_skill.examples.demo_random_action -e "ReplicaCAD_SceneManipulation-v1" --render-mode="human" --shader="rt-fast"
```
It reports the following error:
```
self.window = RenderWindow(*resolution, self.shader_dir)
RuntimeError: Create window failed: Renderer does not support display.
```
I checked the vulkaninfo and its output information is normal.

StoneT2000, 2024-07-03:
There may be a solution using x-forwarding or VNC but it probably will not work / be very slow unfortunately.

closed 2024-07-03 by DevinQiao

lozoRUST, 2024-09-29:
> 我在远程 linux 服务器中安装 maniskill3，并使用以下代码运行示例环境 ReplicaCAD：
> 
> ```
> python -m mani_skill.examples.demo_random_action -e "ReplicaCAD_SceneManipulation-v1" --render-mode="human" --shader="rt-fast"
> ```
> 
> 它报告以下错误：
> 
> ```
> self.window = RenderWindow(*resolution, self.shader_dir)
> RuntimeError: Create window failed: Renderer does not support display.
> ```
> 
> 我检查了 vulkaninfo，它的输出信息是正常的。

I‘ve comed accross the same problem. Have you found the solution?

StoneT2000, 2024-09-30:
@lozoRUST Do you have a display monitor? You must use a display monitor to view the UI

lozoRUST, 2024-09-30:
@StoneT2000 Thanks for reply! Yes, i've tried xfec and saw the desktop on VNC succefully. But the terminal still domonstrated the same RuntimeError: Create window failed: Renderer does not support display.

yizt, 2024-10-16:
> @StoneT2000 Thanks for reply! Yes, i've tried xfec and saw the desktop on VNC succefully. But the terminal still domonstrated the same RuntimeError: Create window failed: Renderer does not support display.

I met the same problem using VNC,did you solve it?

junjzhang, 2025-01-07:
Hi, if any one met this issue. You could solve this using turbovnc + virutalGL

SoshinK, 2025-02-20:
I try to use turbovnc + virutalGL, but have the following error:

```
RuntimeError: vk::Device::createSwapchainKHRUnique: ErrorOutOfDeviceMemory
```

I think it can be also related to https://github.com/haosulab/ManiSkill/issues/680