mani-skill/ManiSkill #406: How to create and control a floting joint?

HeegerGao, 2024-07-06:
Hi, I am using ManiSkill for dexterous hand manipulation tasks. I want to create a "flying-hand" environment where the base link of the dexterous hand can be controlled in 6-DOF, rather than fixed in a position. My problem is that I find that in the [SAPIEN document](https://sapien.ucsd.edu/docs/latest/tutorial/basic/create_articulations.html), it only supports three types of joints: fixed, revolute, and prismatic. 

How to add a floating joint, and how to implement the corresponding controller for such a joint? Do I need to add 6 joints (3 for revolute and 3 for prismatic) for this function? Do you have any sample code? Thanks!

StoneT2000, 2024-07-07:
Yep you exactly do that. Create 6 fake joints to control all 3. Any order should work. I can publish an official floating panda gripper example next week.

HeegerGao, 2024-07-08:
Thanks. Looking forward to the example code.

HeegerGao, 2024-07-11:
> Yep you exactly do that. Create 6 fake joints to control XYZ RPY in the URDF as parents of the base link.. Any order should work. I can publish an official floating panda gripper example next week.

Hi, any progress on this? Thanks.

StoneT2000, 2024-07-12:
Will try and upload an example tomorrow / saturday

StoneT2000, 2024-07-17:
uploaded, pull from the latest git commit to get access

closed 2024-07-17 by StoneT2000

SparKgod1, 2024-07-23:
@HeegerGao Could you share your flying-hand URDF file? Thank you very much.

HeegerGao, 2024-07-23:
> @HeegerGao Could you share your flying-hand URDF file? Thank you very much.

You can refer to the URDF in the official example: https://github.com/haosulab/ManiSkill/blob/main/mani_skill/assets/robots/panda/panda_v2_gripper.urdf

SparKgod1, 2024-07-23:
> > @HeegerGao Could you share your flying-hand URDF file? Thank you very much.
> 
> You can refer to the URDF in the official example: https://github.com/haosulab/ManiSkill/blob/main/mani_skill/assets/robots/panda/panda_v2_gripper.urdf

It seems that directly editing the URDF file to create 6 fake joints is not feasible. How did you modify the URDF file for the dexterous hand like allegro? Thank you for your response

SparKgod1, 2024-07-23:
> It seems that directly editing the URDF file to create 6 fake joints is not feasible. How did you modify the URDF file for the dexterous hand like allegro? Thank you for your response

Or it's feasible I directly edit the URDF file to add the six-joint chain?I hope to receive your guidance, thank you.