mani-skill/ManiSkill #408: How to take pictures from different cameras?

wangyancongAC, 2024-07-07:
I want to set up multiple cameras in the scene，and I set the various parameters of these cameras. They will take pictures every 0.1 seconds and store them locally. Note that the pictures taken by different cameras must be separate, not a large picture stitched together. How should I do this?

StoneT2000, 2024-07-07:
They currently work this way already and are not one large image stitched together. To do this see https://maniskill.readthedocs.io/en/latest/user_guide/tutorials/custom_tasks/intro.html#optional-setting-up-cameras-sensors-for-observations-and-recording for how to add cameras to a task and define them. Note the sensor related properties accept lists of camera configurations to create multiple cameras.

To then take a picture you can call `env.get_obs()` to get the environment observation, which returns data in predefined format. See https://maniskill.readthedocs.io/en/latest/user_guide/concepts/observation.html for what formats are available.

StoneT2000, 2024-07-17:
By default env.render_sensors will return an rgb array of all sensor data as images (rgb, depth by default). You can check that code to see how the images are generated. https://github.com/haosulab/ManiSkill/blob/c1a2c298b81f78672c79989c9218ddde63521934/mani_skill/envs/sapien_env.py#L1106

To make a video, given a saved sequence of images you can use

```bash
from mani_skill.utils.visualization.misc import images_to_video
images_to_video(imgs, output_dir, video_name=file_name, fps=30)
```

wangyancongAC, 2024-07-20:
> By default env.render_sensors will return an rgb array of all sensor data as images (rgb, depth by default). You can check that code to see how the images are generated.
> 
> https://github.com/haosulab/ManiSkill/blob/c1a2c298b81f78672c79989c9218ddde63521934/mani_skill/envs/sapien_env.py#L1106
> 
> To make a video, given a saved sequence of images you can use
> 
> ```shell
> from mani_skill.utils.visualization.misc import images_to_video
> images_to_video(imgs, output_dir, video_name=file_name, fps=30)
> ```

wangyancongAC, 2024-07-20:
> By default env.render_sensors will return an rgb array of all sensor data as images (rgb, depth by default). You can check that code to see how the images are generated. [...]

Is there a way to get the video directly from each camera? Or is your method essentially the same as getting the video directly?

By the way, I also tried 3 other methods.
The first is to use --record-dir="videos", but this method has green text information on the top of the video screen, which I don't like;
The second is to use --save-video, but this method will stitch the screens of multiple envs into the video screen instead of getting multiple different videos;
The third is to use RecordEpisode wrapper, but when I set up multiple cameras, their screens will be stitched together instead of being divided into separate videos.

Is the one you mentioned of combining the image sequences from each camera into videos the best approach?

StoneT2000, 2024-07-20:
To get individual camera images / videos you have to write some code yourself to save the sequence of images from each camera and create separate videos.

There is no way to directly generate a video, the process is always generating an image sequence first then generating the video. I'm sure there is some video streaming solution but I doubt anyone has implemented that for simulators.

wangyancongAC, 2024-07-21:
OK, thanks a lot!

closed 2024-07-22 by StoneT2000