mani-skill/ManiSkill #409: Record trajectories ray-tracing

alexcbb, 2024-07-09:
Hello,

First of all, thank you for you amazing work on the framework.

I'm currently trying to implement new tasks for my own project using Maniskill and thus inherited a new BaseEnv class.
The idea would then to gather trajectories from a random policy in this env class but I want to save the observations as "ray-traced" images (because I've setup some object with metal like material). 
I followed mainly the [demo_random_action](https://github.com/haosulab/ManiSkill/blob/main/mani_skill/examples/demo_random_action.py ) example but was only able to get classical images without ray-tracing. When I set the shader to "rt-fast" I indeed get the following warning/error : "RuntimeError: Currently you cannot use ray-tracing while running simulation with visual observation modes. You may still use
                env.render_rgb_array() or the RecordEpisode wrapper to save videos of ray-traced results" but I do not find any way to do it with RecordEpisode and the env.render_rgb_array() also only provide flat images without real ray-tracing . Do you have some clue on how I could do that ?

Thank you in advance !

StoneT2000, 2024-07-09:
How are you creating environments? For ray traced results you need to ensure obs_mode is not set to any visual observation modes like rgbd. Then RecordEpisode wrapper will work

alexcbb, 2024-07-10:
Here is how I create my env : 
` # Create the environment
    env: BaseEnv  = gym.make(
        args.task_name,
        obs_mode=args.obs_mode,
        control_mode=args.control_mode,
        render_mode=args.render_mode,
        shader_dir=args.shader_dir,
        num_envs=args.num_envs,
        sim_backend=args.sim_backend,
        enable_shadow=True
    )
`

With the following args :
task_name: "PushShape"
num_envs: 1
obs_mode: "rgb" 
control_mode: "pd_joint_delta_pos"
render_mode: "human"
pause: true
sim_backend:  "gpu" 
shader_dir: "rt-fast"
video_dir: "./video"
steps_per_video: 128
random_policy: true
seed: null
verbose: false

And indeed I have a visual observation. But what I don't understand is : will the trajectory recorded by RecordEpisode contain images or not ? If I set the obs to "state", the obs won't be images right ? I indeed would like to have trajectories containing the ray-traced images in order to perform Visual-Based RL.

Thank you again for your reactivity

StoneT2000, 2024-07-10:
Observations of the environment are independent of rendered data for video recordings usually.

If you want to use a vision based policy to take actions, you will need to create a separate "twin" environment used exclusively for rendering at the moment. In this twin environment you create it the same exact way you do with the eval environment that has `obs_mode="rgb"`, but in this one you set `obs_mode="none"` and set `shader_dir="rt-fast"`. 

Then you can do this

```python
eval_env = gym.make(..., obs_mode="rgb")
eval_env_twin = gym.make(..., obs_mode="none", shader_dir="rt-fast")
eval_env_twin = RecordEpisode(eval_env_twin, ....) # only record episode for the twin env
obs, _ = eval_env.reset(seed=0)
eval_env_twin.reset(seed=0)
for i in range (1000):
    obs, reward, terminated, truncated, info = eval_env.step(agent.get_action(obs))
    eval_env_twin.set_state_dict(eval_env.get_state_dict())
```

alexcbb, 2024-07-11:
Ok, I will try it this way and let you know whether it follows my requirements. Thank you again for your answer !

alexcbb, 2024-07-12:
I tried the code and it unfortunately do not register any videos. I checked at the `RecordEpisode` code and saw that there is the following method : 
```
def capture_image(self):
        img = self.env.render()
        img = common.to_numpy(img)
        if len(img.shape) > 3:
            img = tile_images(img, nrows=self.video_nrows)
        return img
```

The problem being that the call of `env.render()` for an env with `rt-fast` shader and `human` render mode will return a `Viewer` from sapien and not an image, so no video will be returned. I didn't find a way to get an image from the Viewer (even in the SAPIENS doc), would you have maybe a clue to help me for this ? Thanks again

StoneT2000, 2024-07-12:
Why is the `human` render mode being used? You should use `render_mode="rgb_array"` instead.

alexcbb, 2024-07-12:
Ok I see, thank you for your help, I think now this is working as I want

closed 2024-07-12 by alexcbb