mani-skill/ManiSkill #417: How to get camera extrinsic and intrinsic parameter?

oym1994, 2024-07-12:
Hi, thanks for your great job. I want to use this data to train a 3D policy where camera intrinsic and extrinsic parameter are needed but unfortunately I cant see this information in the raw data. So can you provide this information or tell how to get these？

Thanks for your attention and wait for your kind response!

StoneT2000, 2024-07-12:
It should in the observations by default. I think it is in obs["sensor_params"]["camera_name"]. Make sure `obs_mode="rgbd"` or one of the visual modes.

oym1994, 2024-07-15:
> It should in the observations by default. I think it is in obs["sensor_params"]["camera_name"]. Make sure `obs_mode="rgbd"` or one of the visual modes.

Thank you! I will try .

StoneT2000, 2024-07-27:
@oym1994 has your issue been resolved?

StoneT2000, 2024-08-01:
Closing the issue now as it has become stale.

closed 2024-08-01 by StoneT2000