mani-skill/ManiSkill #418: How to get all environment IDs of ReplicaCAD and AI2THOR?

fujiawei0724, 2024-07-12:
Great simulator! I have a question about how to get all environment IDs provided by ReplicaCAD and AI2THOR? Looking forward to your reply

StoneT2000, 2024-07-12:
Our full scene environments are currently just for demo/trial. We will have a mobile-manipulation suite of tasks for these full scenes in the future similar to habitat rearrange challenge.

For now you can explore them / take random actions in the envs `"ReplicaCAD_SceneManipulation-v1"` and `"ArchitecTHOR_SceneManipulation-v1"`

See https://maniskill.readthedocs.io/en/latest/user_guide/datasets/scenes.html#exploring-the-scene-datasets for more details

fujiawei0724, 2024-07-12:
Thanks for your prompt help. Another question is how to get the camera and segmentation information in ReplicaCAD_SceneManipulation-v1 and ArchitecTHOR_SceneManipulation-v1? Looking forward to your reply, thanks again!

StoneT2000, 2024-07-12:
When you create the environment, if you obs_mode="rgbd", follow https://maniskill.readthedocs.io/en/latest/user_guide/concepts/observation.html to see how observations are structured (it includes camera matrices, segmentation etc.)

fujiawei0724, 2024-07-13:
Thanks for your always prompt help. Your help helped me learn this simulator much faster. My another question is how to customize ReplicaCAD_SceneManipulation-v1 and ArchitecTHOR_SceneManipulation-v1? I want to modify them to match my development conditions, but now I cannot access any class or config defining them. Could you please help me figure out how to modify them by myself and where is the class or config that define the mobile manipulation environment (ReplicaCAD_SceneManipulation-v1 and ArchitecTHOR_SceneManipulation-v1)? Thanks for your help.

StoneT2000, 2024-07-13:
You can start by copying the code here: https://github.com/haosulab/ManiSkill/blob/main/mani_skill/envs/scenes/base_env.py

Customizing this might not be super easy since we don't have complete documentation on how to build and manage more complex scenes. In particular this code supports any scene from different datasets provided a scene builder is implemented (follow the imports to see how the code is defined for scene builders)

fujiawei0724, 2024-07-14:
Thanks for your answer, I have learned more about the simulator. Great work and help!

fujiawei0724, 2024-07-15:
My another question is: is there any API to crop the panoramic point cloud conveniently in the mobile manipulation scenes? I have noticed that the obs supports point cloud mode and I can access the point cloud collected by the robot in its position. Could you please help me figure is there any method I can get the panoramic point cloud of the whole scene (but not collected from the current position)? Thanks again.

StoneT2000, 2024-07-15:
What does panoramic refer to here? If you want a pointcloud of the whole scene you would need to get the visual mesh of every object (turn them into trimesh objects) and sample points on the surfaces and combine them all in the world frame.

fujiawei0724, 2024-07-16:
I am sorry for the confusion. You are right, I want to get the point cloud of the whole scene, if there any convenient API in Sapien to get them? Looking forward ti your reply.

fujiawei0724, 2024-07-16:
I have solved the problem following your guidance, thanks very much!

fujiawei0724, 2024-07-18:
I am sorry to open this issue again. Following your guidance, it is easy to crop the point clouds of all the objects in the scene, but it seems there is no point cloud for the walls (the room), is there any method to crop the point cloud of the walls (room) in the environment (ReplicaCAD and AI2THOR)? Looking forward to your reply~

StoneT2000, 2024-07-18:
Is the background not an actor object? I don't remember what jt was named but it should be one of the actors and has a collision mesh / visual mesh you can sample from.
m

Ethan-Chen-plus, 2025-05-16:
> I am sorry for the confusion. You are right, I want to get the point cloud of the whole scene, if there any convenient API in Sapien to get them? Looking forward ti your reply.
> I have solved the problem following your guidance, thanks very much!

Hi!@fujiawei0724 Could you please tell me how to get the point cloud of the whole scene? I have stuck into this problems these days. Thanks for your helping!