mani-skill/ManiSkill #423: Articulation object - get_net_contact_forces with multiple link names

rechim25, 2024-07-15:
I am trying to get the contact forces for each finger of the Franka robot in the PickCube-v1 environment on the CPU. To achieve this I 'm using the get_net_contact_forces from the Articulation class as follows:

contact_forces = env.unwrapped.agent.robot.get_net_contact_forces(link_names=['panda_leftfinger', 'panda_rightfinger'])
print(contact_forces)

The returned tensor is of shape (1, 3):  tensor([[ 0.0662, -0.0682,  0.3219]])

However, the documentation of the function says we should expect that get_net_contact_forces "Returns force vector of shape (N, len(link_names), 3) where N is the number of environments" (https://github.com/haosulab/ManiSkill/pull/354/commits/e1bcd5efbad3fe9f2906b2a61846af74741bb236).

I have checked and both links 'panda_leftfinger' and 'panda_rightfinger' exist on the robot.

To avoid the issue above I call get_net_contact_forces two times for each individual link but I suspect this may be slower:
left_contact_forces = self.env.unwrapped.agent.robot.get_net_contact_forces(link_names=['panda_leftfinger'])
right_contact_forces = self.env.unwrapped.agent.robot.get_net_contact_forces(link_names=['panda_rightfinger'])

Just wondering if I'm missing something about how get_net_contact_forces should behave or if this is a bug?

StoneT2000, 2024-07-15:
Can you share your full environment code? I get the right shape of data in GPU simulation e.g. (1024, 2, 3) for your code with 1024 parallel envs.

StoneT2000, 2024-07-27:
Hi @rechim25 is this still an issue?

StoneT2000, 2024-08-01:
Closing the issue now as it has become stale.

rechim25, 2024-09-10:
Hi, apologies for the slow reply. It still seems to be an issue.

The issue is on the CPU with Panda Arm when I try to get contact forces for fingers. When calling to get forces for both fingers I get a tensor of shape [1,3] whereas it should be [1, 2, 3] from my understanding (i.e., [num_envs, len(link_names), 3]).

This is an example code where the issue happens:
```
import gymnasium as gym
import numpy as np
import time
from mani_skill.envs.sapien_env import BaseEnv

if __name__ == '__main__':
    env: BaseEnv = gym.make(
        id='PickCube-v1',
        num_envs=1,
        obs_mode='pointcloud',
        control_mode='pd_joint_pos',
        render_mode='human',
        reward_mode='sparse',
        enable_shadow=True,
        sim_backend='cpu',
    )

    env.reset(seed=0)

    for i in range(5000):
        random_action = np.random.rand(8)
        obs, _, _, _, _ = env.step(action=random_action)
        forces = env.agent.robot.get_net_contact_forces(
            link_names=['panda_rightfinger', 'panda_leftfinger']
        )  # (n_envs, n_links, 3)???
        env.render_human()
        time.sleep(0.1)
        print(forces.shape)

```

StoneT2000, 2024-09-10:
Should be fixed now @rechim25, thanks for spotting the bug

PR #556 (2024-09-10): [BugFix] Fix CPU articulation net contact forces being wrong shape
Closes #423
closed 2024-09-10 by StoneT2000