mani-skill/ManiSkill #425: The ManiSkill3 is not supporting DAPG?

SparKgod1, 2024-07-16:
I want to train DAPG on dexycb dataset,should i use ManiSkill2 rather than ManiSkill3?I only find DAPG in ManiSkill2-Learn,appreciate your reply!

StoneT2000, 2024-07-16:
We can add back a DAPG baseline but it might take some time. We are currently prioritizing adding a [RLPD](https://github.com/ikostrikov/rlpd) baseline and have already added the [RFCL](https://github.com/StoneT2000/rfcl) baseline which are both in the line of online imitation learning methods like DAPG (and run faster probably).

SparKgod1, 2024-07-17:
Thank you for your reply. I appreciate the efforts being made to add more baselines. I will explore  RFCL.

StoneT2000, 2024-07-18:
The RLPD baseline has been added. It is quite sample efficient (when given sufficient, not too multi-modal, demonstrations). Might struggle more than RFCL if you don't use enough demonstrations / use hard to learn from demos. (RFCL leverages env state reset to accelerate learning, RLPD does not).

SparKgod1, 2024-07-18:
> The RLPD baseline has been added in the latest git commit of this repo. It is quite sample efficient (when given sufficient, not too multi-modal, demonstrations). Might struggle more than RFCL if you don't use enough demonstrations / use hard to learn from demos. (RFCL leverages env state reset to accelerate learning, RLPD does not).

Thank you so much for the information and your hard work! I will carefully evaluate the performance of RLPD and RFCL and choose the most suitable baseline for my needs. Your point about RLPD's requirements for the number and diversity of demonstrations is particularly helpful to me. Thanks again!