mani-skill/ManiSkill #426: Segmentation fault (core dumped)

baiyu858, 2024-07-16:
`python -m mani_skill.examples.demo_random_action -e PickCube-v1`
output:
`opts: []
env_kwargs: {}
Segmentation fault (core dumped)
`
GPU:RTX 3090

StoneT2000, 2024-07-16:
Can you share the result of nvidia-smi, pip list, and what linux system you are on?

baiyu858, 2024-07-17:
> Can you share the result of nvidia-smi, pip list, and what linux system you are on?
Thank you for your reply.
I had this problem at the beginning of the step, I don't know what is the reason? mani-skill2 has the same problem.
I'm using Ubuntu 20.04 LTS on the server side (GNU/Linux 5.4.0-174-generic x86_64).
nvidia-smi:
`    
+-----------------------------------------------------------------------------------------+
| NVIDIA-SMI 550.54.15              Driver Version: 550.54.15      CUDA Version: 12.4     |
|-----------------------------------------+------------------------+----------------------+
| GPU  Name                 Persistence-M | Bus-Id          Disp.A | Volatile Uncorr. ECC |
| Fan  Temp   Perf          Pwr:Usage/Cap |           Memory-Usage | GPU-Util  Compute M. |
|                                         |                        |               MIG M. |
|=========================================+========================+======================|
|   0  NVIDIA GeForce RTX 3090        Off |   00000000:1A:00.0 Off |                  N/A |
| 30%   31C    P8             31W /  350W |      14MiB /  24576MiB |      0%      Default |
|                                         |                        |                  N/A |
+-----------------------------------------+------------------------+----------------------+
|   1  NVIDIA GeForce RTX 3090        Off |   00000000:1B:00.0 Off |                  N/A |
| 30%   30C    P8             20W /  350W |      14MiB /  24576MiB |      0%      Default |
|                                         |                        |                  N/A |
+-----------------------------------------+------------------------+----------------------+
|   2  NVIDIA GeForce RTX 3090        Off |   00000000:3D:00.0 Off |                  N/A |
| 30%   27C    P8             18W /  350W |      14MiB /  24576MiB |      0%      Default |
|                                         |                        |                  N/A |
+-----------------------------------------+------------------------+----------------------+
|   3  NVIDIA GeForce RTX 3090        Off |   00000000:3E:00.0 Off |                  N/A |
| 30%   29C    P8             21W /  350W |      14MiB /  24576MiB |      0%      Default |
|                                         |                        |                  N/A |
+-----------------------------------------+------------------------+----------------------+
|   4  NVIDIA GeForce RTX 3090        Off |   00000000:88:00.0 Off |                  N/A |
| 65%   58C    P2            229W /  350W |   15461MiB /  24576MiB |    100%      Default |
|                                         |                        |                  N/A |
+-----------------------------------------+------------------------+----------------------+
|   5  NVIDIA GeForce RTX 3090        Off |   00000000:89:00.0 Off |                  N/A |
|  0%   52C    P2            226W /  350W |   15939MiB /  24576MiB |    100%      Default |
|                                         |                        |                  N/A |
+-----------------------------------------+------------------------+----------------------+
|   6  NVIDIA GeForce RTX 3090        Off |   00000000:B1:00.0 Off |                  N/A |
| 51%   52C    P2            229W /  350W |   16203MiB /  24576MiB |    100%      Default |
|                                         |                        |                  N/A |
+-----------------------------------------+------------------------+----------------------+
|   7  NVIDIA GeForce RTX 3090        Off |   00000000:B2:00.0 Off |                  N/A |
| 52%   51C    P2            230W /  350W |   15839MiB /  24576MiB |     89%      Default |
|                                         |                        |                  N/A |
+-----------------------------------------+------------------------+----------------------+
                                                                                         
+-----------------------------------------------------------------------------------------+
| Processes:                                                                              |
|  GPU   GI   CI        PID   Type   Process name                              GPU Memory |
|        ID   ID                                                               Usage      |
|=========================================================================================|
|    0   N/A  N/A   3955805      G   /usr/lib/xorg/Xorg                              8MiB |
|    1   N/A  N/A   3955805      G   /usr/lib/xorg/Xorg                              8MiB |
|    2   N/A  N/A   3955805      G   /usr/lib/xorg/Xorg                              8MiB |
|    3   N/A  N/A   3955805      G   /usr/lib/xorg/Xorg                              8MiB |
|    4   N/A  N/A   1900905      C   ...gzhou/anaconda3/envs/tc2/bin/python       6584MiB |
|    4   N/A  N/A   2032157      C   ...gzhou/anaconda3/envs/tc2/bin/python       8850MiB |
|    4   N/A  N/A   3955805      G   /usr/lib/xorg/Xorg                              8MiB |
|    5   N/A  N/A   1900906      C   ...gzhou/anaconda3/envs/tc2/bin/python       6550MiB |
|    5   N/A  N/A   2032158      C   ...gzhou/anaconda3/envs/tc2/bin/python       9362MiB |
|    5   N/A  N/A   3955805      G   /usr/lib/xorg/Xorg                              8MiB |
|    6   N/A  N/A   1900907      C   ...gzhou/anaconda3/envs/tc2/bin/python       6982MiB |
[... cut ...]
tabulate                 0.9.0
toppra                   0.6.0
torch                    2.3.1
torchaudio               2.3.1
torchvision              0.18.1
tqdm                     4.66.4
traitlets                5.14.3
transforms3d             0.4.2
trimesh                  4.4.3
triton                   2.3.1
typing_extensions        4.12.2
tyro                     0.8.5
urllib3                  2.2.2
wcwidth                  0.2.13
wheel                    0.43.0
zipp                     3.19.2
`

StoneT2000, 2024-07-17:
I hope the bug is not related to having 8 connected GPUs (we have not tested on such a setup before). 

A few possible ideas:
- Install a older version of numpy (1.26.4 works)
- Install a older version of torch (2.2.1 works)

Just realized with the new numpy release I need to pin our numpy dependency to <2.0.0 probably.

If the two other installs do not fix it, there may be a chance your nvidia driver is too new (550). 535.183.01 and cuda version 12.2 is what I run now (535.X should work). You can try downgrading the driver (if you use apt, make sure you install nvidia-driver-xxx without the word server or headless in its name).

Can you also run `ldconfig -p | grep libGLX_nvidia`?

baiyu858, 2024-07-19:
`ldconfig -p | grep libGLX_nvidia`
output:
libGLX_nvidia.so.0 (libc6,x86-64) => /lib/x86_64-linux-gnu/libGLX_nvidia.so.0

I tried to lower the numpy and torch versions, but it didn't work, it was still a Segmentation fault (core dumped). Has anyone else had this problem? I was afraid to drop the driver version because others were sharing the server. What is the fault caused by this [Segmentation fault (core dumped)], which is strange, and nothing else is reported?

StoneT2000, 2024-07-22:
@fbxiang any idea? The driver version is very new and I have not used it before so I suspect it might be that.

StoneT2000, 2024-07-22:
Actually @baiyu858 can you run

```
python -m sapien.example.hello_world
```

baiyu858, 2024-07-23:
> Actually @baiyu858 can you run 你能跑吗?
> 
> ```
> python -m sapien.example.hello_world
> ```

`python -m sapien.example.hello_world`
output:
Segmentation fault (core dumped)

StoneT2000, 2024-07-24:
Another option is to try and pull the docker image and run the example code from there? You will need to setup nvidia docker containers if you haven't done so before. https://maniskill.readthedocs.io/en/latest/user_guide/getting_started/docker.html

tongzhoumu, 2024-07-24:
@baiyu858 try to downgrade your numpy to 1.26

fbxiang, 2024-07-24:
> @baiyu858 try to downgrade your numpy to 1.26

This makes a lot of sense.

StoneT2000, 2024-07-25:
I asked the user to downgrade earlier to 1.26 but they said it did not work

StoneT2000, 2024-07-25:
try reinstalling vulcan, and also follow the vulcan trouble shooting guide on our docs (make sure those icd files are there)

StoneT2000, 2024-08-20:
Closing the issue as it is stale

closed 2024-08-20 by StoneT2000

lieyanzhan, 2025-12-18:
> Actually [@baiyu858](https://github.com/baiyu858) can you run
> 
> ```
> python -m sapien.example.hello_world
> ```

> Can you share the result of nvidia-smi, pip list, and what linux system you are on?

[image]
l  have the same problem,  and I run the maniskill in the docker, how can l fix this problem?