mani-skill/ManiSkill #429: from_pd_joint_delta_pos doesn't work

sean1295, 2024-07-16:
In replay_trajectory.py, when trying to convert environment in pd_joint_delta_pos to pd_joint_pos using the following commands,

python -m mani_skill.trajectory.replay_trajectory --traj-path demo/StackCubeEnv/20240716_102711.h5 --save-traj -o rgb --target-control-mode pd_ee_delta_pose --num-procs 4 --use-first-env-state  --max-retry 5

it keeps giving me following messages:
"Episode xxx is not replayed successfully. Skipping."

I also kept track of the joint positions of the robots across two environments using the converted actions and they do not follow exactly. Please see below. 
[image]

Was looking at the code but do not see anything that stands out. aside from one error inside clip_and_scale_action function where clip method from numpy should be used instead of torch.

Am wondering if there are ways to fix this.

StoneT2000, 2024-07-16:
The problem of 100% accurate action space conversion is highly non-trivial (if you solve it it's worth a conference publication). It is quite difficult to ensure the resulting environment state from using a different controller is the same as the original trajectory data, we can only do approximations and at best, try to get a success rate that matches the trajectory data

sean1295, 2024-07-16:
Hi Stone, thanks for your prompt reply. I agree. When trying to convert from pd_joint_pos to pd_ee_delta_pose, I noticed that ~1% of them didn't convert successfully, which I am totally fine with since I can simply generate more trajectories from original env. But this seemly easy conversion from pd_joint_delta_pos  to pd_joint_pos actually has 0 % conversion rate. I think this particular conversion has a lot of utility since many RL agents can be trained with pd_joint_delta_pos and we may want to convert that to other action spaces. In order to do so, we first need to convert that back to pd_joint_pos, and then the target action space.

StoneT2000, 2024-07-16:
If it has 0 then that sounds like a bug. I'll keep the issue up and take a long soon.

lindicaphxag-tech, 2026-10-06:
I traced the `pd_joint_delta_pos -> pd_joint_pos` path and found a controller-semantics mismatch that may explain the 0% conversion rate.

The source action is a normalized delta-joint command, so it first needs to be decoded into physical Δq. The resulting physical target qpos then needs to be encoded again in the target `PDJointPosController` action chart. The current path appears to pass physical qpos directly to a normalized target controller, which can cause it to be scaled again.

I prepared a focused patch and end-to-end regression test that make the conversion explicit:

normalized source action → physical Δq → physical target qpos → target-controller native action.

For example, starting from qpos `[0.2, -0.3]`, a normalized delta action `[0.5, -0.5]` under ±0.1 physical delta limits should produce the absolute target `[0.25, -0.35]`.

The patch also preserves clipping/representability behavior instead of silently treating an out-of-range target as an exact conversion.

I have a minimal two-file PR-ready branch with focused CI passing. If this matches the intended controller semantics, I can open the PR against #429.