mani-skill/ManiSkill #434: Robot control in ArchitecTHOR_SceneManipulation-v1

fujiawei0724, 2024-07-17:
Thanks for the great platform, and you have solved lots of problems for me, thanks for the patient help. 

In ArchitecTHOR_SceneManipulation-v1, is there a way to quickly control the robot so that it can move to a specified location or grab an object at a specified location?

Looking forward to your reply. Thanks again.

StoneT2000, 2024-07-17:
That's a pretty hard problem! We are currently working a RL approach to learn contact rich long horizon manipulation but it's not trivial.

We will eventually integrate some teleoperation solutions but they aren't a priority at this moment.

StoneT2000, 2024-07-18:
Ah this is due to how free joints / mobile bases are modeled in simulation. Mobile bases are modeled by three joints, X, Y prismatic and a resolute joint along the z axis. So the first 3 qpos values of the robot give you the x, y position and z axis rotation of the robot relative to the pose.

fujiawei0724, 2024-07-19:
Thank you very much~

closed 2024-07-19 by fujiawei0724