mani-skill/ManiSkill #443: RuntimeError: vk::createInstanceUnique: ErrorOutOfHostMemory

ShiguangSun, 2024-07-22:
Hi, I just installed Maniskill2 on Ubuntu 16.04 with pip install mani-skill2==0.4.2. I tried an experiment and I got following error:

[2024-07-22 17:15:10.714] [svulkan2] [error] createInstance: present requested but GLFW does not support Vulkan. Continue without GLFW.
Traceback (most recent call last):
  File "experiments/train_repo.py", line 120, in <module>
    env = make_env(config.env_id, config.seed, config.pixel_obs)
  File "/home/cxg19/ws/rl/sim_repo/environments/__init__.py", line 92, in make_env
    env = gym.make(
  File "/home/cxg19/anaconda3/envs/tdmpc/lib/python3.8/site-packages/gym/envs/registration.py", line 235, in make
    return registry.make(id, **kwargs)
  File "/home/cxg19/anaconda3/envs/tdmpc/lib/python3.8/site-packages/gym/envs/registration.py", line 129, in make
    env = spec.make(**kwargs)
  File "/home/cxg19/anaconda3/envs/tdmpc/lib/python3.8/site-packages/gym/envs/registration.py", line 87, in make
    env = self.entry_point(**_kwargs)
  File "/home/cxg19/anaconda3/envs/tdmpc/lib/python3.8/site-packages/mani_skill2/utils/registration.py", line 92, in make
    env = env_spec.make(**kwargs)
  File "/home/cxg19/anaconda3/envs/tdmpc/lib/python3.8/site-packages/mani_skill2/utils/registration.py", line 34, in make
    return self.cls(**_kwargs)
  File "/home/cxg19/anaconda3/envs/tdmpc/lib/python3.8/site-packages/mani_skill2/envs/pick_and_place/pick_cube.py", line 22, in __init__
    super().__init__(*args, **kwargs)
  File "/home/cxg19/anaconda3/envs/tdmpc/lib/python3.8/site-packages/mani_skill2/envs/pick_and_place/base_env.py", line 27, in __init__
    super().__init__(*args, **kwargs)
  File "/home/cxg19/anaconda3/envs/tdmpc/lib/python3.8/site-packages/mani_skill2/envs/sapien_env.py", line 102, in __init__
    self._renderer = sapien.SapienRenderer(**renderer_kwargs)
RuntimeError: vk::createInstanceUnique: ErrorOutOfHostMemory

I have done the Vulkan thing as suggested in the Troubleshooting section https://maniskill.readthedocs.io/en/latest/user_guide/getting_started/installation.html#troubleshooting.  There was not the file /usr/share/vulkan/icd.d/nvidia_icd.json. Then I created the /usr/share/vulkan/icd.d/nvidia_icd.json. But the file /usr/share/glvnd/egl_vendor.d/10_nvidia.json and the file /etc/vulkan/implicit_layer.d/nvidia_layers.json exist. However, the "api_version" in the suggested document is "1.2.155", while the "api_version" in my /etc/vulkan/implicit_layer.d/nvidia_layers.json file is "1.3.277". Does this have any impact? I tried modifying them to be consistent, but it still gives an error.

StoneT2000, 2024-07-22:
The error you received means you ran out of GPU memory to rendering somehow. 

Can you run
```
nvidia-smi
ldconfig -p | grep libGLX_nvidia
```

ShiguangSun, 2024-07-23:
nvidia-smi
outputs：
+-----------------------------------------------------------------------------------------+
| NVIDIA-SMI 550.78                 Driver Version: 550.78         CUDA Version: 12.4     |
|-----------------------------------------+------------------------+----------------------+
| GPU  Name                 Persistence-M | Bus-Id          Disp.A | Volatile Uncorr. ECC |
| Fan  Temp   Perf          Pwr:Usage/Cap |           Memory-Usage | GPU-Util  Compute M. |
|                                         |                        |               MIG M. |
|=========================================+========================+======================|
|   0  NVIDIA GeForce RTX 2080 Ti     Off |   00000000:04:00.0 Off |                  N/A |
| 27%   30C    P8             25W /  250W |      22MiB /  11264MiB |      0%      Default |
|                                         |                        |                  N/A |
+-----------------------------------------+------------------------+----------------------+
|   1  NVIDIA GeForce RTX 2080 Ti     Off |   00000000:05:00.0 Off |                  N/A |
| 27%   28C    P8              3W /  250W |       4MiB /  11264MiB |      0%      Default |
|                                         |                        |                  N/A |
+-----------------------------------------+------------------------+----------------------+
|   2  NVIDIA GeForce RTX 2080 Ti     Off |   00000000:08:00.0 Off |                  N/A |
| 27%   30C    P8             39W /  250W |       4MiB /  11264MiB |      0%      Default |
|                                         |                        |                  N/A |
+-----------------------------------------+------------------------+----------------------+
|   3  NVIDIA GeForce RTX 2080 Ti     Off |   00000000:09:00.0 Off |                  N/A |
| 27%   28C    P8             10W /  250W |       4MiB /  11264MiB |      0%      Default |
|                                         |                        |                  N/A |
+-----------------------------------------+------------------------+----------------------+
|   4  NVIDIA GeForce RTX 2080 Ti     Off |   00000000:85:00.0 Off |                  N/A |
| 27%   28C    P8             13W /  250W |       4MiB /  11264MiB |      0%      Default |
|                                         |                        |                  N/A |
+-----------------------------------------+------------------------+----------------------+
|   5  NVIDIA GeForce RTX 2080 Ti     Off |   00000000:86:00.0 Off |                  N/A |
| 27%   27C    P8              1W /  250W |       4MiB /  11264MiB |      0%      Default |
|                                         |                        |                  N/A |
+-----------------------------------------+------------------------+----------------------+
|   6  NVIDIA GeForce RTX 2080 Ti     Off |   00000000:89:00.0 Off |                  N/A |
| 27%   28C    P8              9W /  250W |       4MiB /  11264MiB |      0%      Default |
|                                         |                        |                  N/A |
+-----------------------------------------+------------------------+----------------------+
|   7  NVIDIA GeForce RTX 2080 Ti     Off |   00000000:8A:00.0 Off |                  N/A |
| 27%   28C    P8              5W /  250W |       4MiB /  11264MiB |      0%      Default |
|                                         |                        |                  N/A |
+-----------------------------------------+------------------------+----------------------+

+-----------------------------------------------------------------------------------------+
| Processes:                                                                              |
|  GPU   GI   CI        PID   Type   Process name                              GPU Memory |
|        ID   ID                                                               Usage      |
|=========================================================================================|
|    0   N/A  N/A      2166      G   /usr/lib/xorg/Xorg                             16MiB |
+-----------------------------------------------------------------------------------------+

ldconfig -p | grep libGLX_nvidia
outputs：
libGLX_nvidia.so.0 (libc6,x86-64) => /usr/lib/x86_64-linux-gnu/libGLX_nvidia.so.0
        libGLX_nvidia.so.0 (libc6) => /usr/lib32/libGLX_nvidia.so.0

I used a completely idle 2080ti GPU and still had this error.

When I ran an example:  python -m mani_skill2.examples.demo_random_action, I still had this error:

opts: []
env_kwargs: {}
[2024-07-23 10:08:51.408] [svulkan2] [error] createInstance: present requested but GLFW does not support Vulkan. Continue without GLFW.
Traceback (most recent call last):
  File "/home/cxg19/anaconda3/envs/tdmpc/lib/python3.8/runpy.py", line 194, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/home/cxg19/anaconda3/envs/tdmpc/lib/python3.8/runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "/home/cxg19/anaconda3/envs/tdmpc/lib/python3.8/site-packages/mani_skill2/examples/demo_random_action.py", line 71, in <module>
    main()
  File "/home/cxg19/anaconda3/envs/tdmpc/lib/python3.8/site-packages/mani_skill2/examples/demo_random_action.py", line 34, in main
    env: BaseEnv = gym.make(
  File "/home/cxg19/anaconda3/envs/tdmpc/lib/python3.8/site-packages/gym/envs/registration.py", line 235, in make
[... cut ...]
  File "/home/cxg19/anaconda3/envs/tdmpc/lib/python3.8/site-packages/mani_skill2/envs/pick_and_place/pick_cube.py", line 22, in __init__
    super().__init__(*args, **kwargs)
  File "/home/cxg19/anaconda3/envs/tdmpc/lib/python3.8/site-packages/mani_skill2/envs/pick_and_place/base_env.py", line 27, in __init__
    super().__init__(*args, **kwargs)
  File "/home/cxg19/anaconda3/envs/tdmpc/lib/python3.8/site-packages/mani_skill2/envs/sapien_env.py", line 102, in __init__
    self._renderer = sapien.SapienRenderer(**renderer_kwargs)
RuntimeError: vk::createInstanceUnique: ErrorOutOfHostMemory

ShiguangSun, 2024-07-23:
> The error you received means you ran out of GPU memory to rendering somehow.
> 
> Can you run
> 
> ```
> nvidia-smi
> ldconfig -p | grep libGLX_nvidia
> ```

However, I am now facing a new issue. I am running an experiment that defaults to using GPU 0. When I use CUDA_VISIBLE_DEVICES to specify a different GPU for other experiments, it gives the following error:

Traceback (most recent call last):
  File "experiments/train_repo.py", line 120, in <module>
    env = make_env(config.env_id, config.seed, config.pixel_obs)
  File "/home/cxg19/ws/rl/sim_repo/environments/__init__.py", line 92, in make_env
    env = gym.make(
  File "/home/cxg19/anaconda3/envs/tdmpc/lib/python3.8/site-packages/gym/envs/registration.py", line 235, in make
    return registry.make(id, **kwargs)
  File "/home/cxg19/anaconda3/envs/tdmpc/lib/python3.8/site-packages/gym/envs/registration.py", line 129, in make
    env = spec.make(**kwargs)
  File "/home/cxg19/anaconda3/envs/tdmpc/lib/python3.8/site-packages/gym/envs/registration.py", line 87, in make
    env = self.entry_point(**_kwargs)
  File "/home/cxg19/anaconda3/envs/tdmpc/lib/python3.8/site-packages/mani_skill2/utils/registration.py", line 92, in make
    env = env_spec.make(**kwargs)
  File "/home/cxg19/anaconda3/envs/tdmpc/lib/python3.8/site-packages/mani_skill2/utils/registration.py", line 34, in make
    return self.cls(**_kwargs)
  File "/home/cxg19/anaconda3/envs/tdmpc/lib/python3.8/site-packages/mani_skill2/envs/pick_and_place/pick_cube.py", line 22, in __init__
    super().__init__(*args, **kwargs)
  File "/home/cxg19/anaconda3/envs/tdmpc/lib/python3.8/site-packages/mani_skill2/envs/pick_and_place/base_env.py", line 27, in __init__
    super().__init__(*args, **kwargs)
  File "/home/cxg19/anaconda3/envs/tdmpc/lib/python3.8/site-packages/mani_skill2/envs/sapien_env.py", line 102, in __init__
    self._renderer = sapien.SapienRenderer(**renderer_kwargs)
RuntimeError: vk::PhysicalDevice::createDeviceUnique: ErrorInitializationFailed

closed 2024-07-23 by StoneT2000