mani-skill/ManiSkill #444: What's the name of YCBs in demostraion

SparKgod1, 2024-07-23:
`
for i, model_id in enumerate(model_ids):
            # TODO: before official release we will finalize a metadata dataclass that these build functions should return.
            builder = actors.get_actor_builder(
                self.scene,
                id=f"ycb:{model_id}",
            )
            builder.set_scene_idxs([i])
            self._objs.append(builder.build(name=f"{model_id}-{i}"))
        self.obj = Actor.merge(self._objs, name="ycb_object")
`
In demonstrations, should I set the name of the YCB object to {model_id}-{i}? What does the i represent in this context?

StoneT2000, 2024-07-23:
What do you mean by in demonstrations should I set object names?

SparKgod1, 2024-07-23:
I want to use my own grasping sequence data as demonstrations to train RFCL. While organizing the data, I noticed that the actors in env_state require actor_id, so I want to find the naming in the builder.

StoneT2000, 2024-07-23:
Ah I remember this particular / potential issue. Env state is not well defined when there might be different objects in the environment. Previously in maniskill 2 all objects had the same name since we only did CPU simulation whereas to support GPU sim they should have different names.

I think I have one potential solution that I will try and add in this week. For now tasks with object geometry randomization might not have useable env states.

StoneT2000, 2024-11-20:
This is now fixed by #676. A state dict registry is now added to support env state parameterization of tasks with object randomizations. In these situations you just need to manually remove the individual actor objects and include only the merged version of them. Already done for PegInsertionSide and multiple other tasks as well.

closed 2024-11-20 by StoneT2000