mani-skill/ManiSkill #447: How to Specify the Rendering GPU?

ShiguangSun, 2024-07-24:
Hello, I have a question. When running an experiment, I specified a GPU, but I found that the GPU I specified seems to be used only for network computations, while the task rendering GPU is still using GPU 0. Is it possible to specify which GPU is used for task rendering?

StoneT2000, 2024-07-24:
At the moment, all simulation and rendering occurs on GPU 0 it seems. I will see if we can get this feature added soon

ShiguangSun, 2024-07-25:
> At the moment, all simulation and rendering occurs on GPU 0 it seems. I will see if we can get this feature added soon

Ok, Thanks for your replay.

StoneT2000, 2024-07-27:
Hi @ShiguangSun do you have a server which supports multiple GPUs and can help test? I have a fix

StoneT2000, 2024-07-27:
You can try the latest code by running

```python
pip install --upgrade git+https://github.com/haosulab/ManiSkill.git
```

and then see the new docs on gpu device selection here:
https://maniskill.readthedocs.io/en/latest/user_guide/getting_started/quickstart.html#additional-gpu-simulation-rendering-customization

PR #455 (2024-07-27): [Feature] SIm/Render device selection
Closes #447
closed 2024-07-27 by StoneT2000