mani-skill/ManiSkill #448: Issue about calculation inverse kinematics for Fetch robot in grasping process

fujiawei0724, 2024-07-24:
Great work and codebase, and I have developed something interesting using this simulator. I face a problem that when I want to use the Fetch robot provided in the simulator to grasp an object given its pose, I don't know how to calculate the inverse kinematics given the target grasping pose. Could you please help me figure out this problem? We can assume the Fetch robot is not required to move the base and only considers the arm to grasp. Thanks~

StoneT2000, 2024-07-24:
You can use the pinnochio library if you want to do IK on the CPU environment. See our pd ee delta pose controller code here for how we use it in the CPU environment: https://github.com/haosulab/ManiSkill/blob/main/mani_skill/agents/controllers/pd_ee_pose.py#L152-L158

You can also try the `mplib` package which is simple to use. There is sample code using it to solve various tasks with the panda arm, you can configure mplib to use fetch

fujiawei0724, 2024-07-31:
> Thanks for your help. I am a little confused about the controller used in Fetch robot, for the arm control in Fetch, it seems that the employed control is named with: PDJointPosController but not PDEEPosController (with IK calculation function itself). I want to figure out the difference between the two types of controllers. Furthermore, if I want to control the arm to target pose in Fetch, whether i need to modify the initial PDJointPosController to PDEEPosController by myself to apply the IK calculation function of it? I am sorry for my naive question about robot control, looking forward to your reply~

I wanted to follow up on the issue I raised a few days ago. I understand that you might be busy, but I would greatly appreciate any updates or feedback you could provide.

Thank you for your time and effort on this project!

StoneT2000, 2024-08-10:
Sorry dropped the ball on this.

The fetch robot has multiple controllers available, you can select which one to use at runtime by setting the control mode when you use gym.make. The difference is joint pos controller controls joints directly, either with absolute or delta actions. The PDEEPos controller uses inverse kinematics to instead control the 3D position of the end effector (the gripper area). 

If you want to control the arm to reach a specific target pose instead of sending delta actions, you can copy the robot class in our code and add a new controller with the same settings as the pd ee delta pose controller but with use_delta=False. (Note that this will only work in CPU sim at the moment, you can see my comment here #488 about support for EE target pose control vs delta EE pose control).

StoneT2000, 2024-08-20:
Closing the issue as it is stale

closed 2024-08-20 by StoneT2000