mani-skill/ManiSkill #449: How to use ManiSkill2 as a benchmark？

yingfei1016, 2024-07-24:
Hello，

I would like to use ManiSkill2 as a benchmark, but **the documentation** has been upgraded to ManiSkill 3.0. I am confused about the use of ManiSkill2, such as how should I **download the data** to align with the previous methods. I see there are multiple versions of the data on Hugging Face, and I am unsure about how to handle the **evaluation**. I hope to maintain consistency with the CVPR workshop！

Thank you very much！

StoneT2000, 2024-07-24:
ManiSkill 3.0 is still in beta and not completely ready to have an official benchmark. It simply is a massive collection of various tasks but the official benchmark (for RL, IL, etc. areas of research) will pick out a subset of tasks and have baseline results for.

For ManiSkill 2, you can use the original code here: https://github.com/haosulab/ManiSkill/tree/v0.5.3 and follow the old documentation there for how to download demos, assets etc.

By default maniskill 3 demos/data will go to the "~/.maniskill" folder while for ManiSkill 2 it will go elsewhere.

If you want to compare against methods that tested on ManiSkill2, my recommendation now is to still use ManiSkill 2, at least until ManiSkill 3 has its comprehensive baseline results completed.

yingfei1016, 2024-07-24:
In  https://github.com/haosulab/ManiSkill/tree/v0.5.3，the document has also updated to latest version. 
[image]

Can you send me a old documentation.

yingfei1016, 2024-07-24:
1、I download the data from https://huggingface.co/datasets/haosulab/ManiSkill2/tree/main/demos/v0/ . However, I find the code in https://github.com/tongzhoumu/ManiSkill_Baselines use a different url path. It is very slow,
[image]

If the https://huggingface.co/datasets/haosulab/ManiSkill2/tree/main/demos/v0/ is the right data path?

2、The document in CVPR workshop also expired.
[image]

StoneT2000, 2024-07-24:
Looks like the link is outdated.

This is the docs link for the maniskill 2: https://maniskill.readthedocs.io/en/v0.5.3-rtd/
For envs: https://maniskill.readthedocs.io/en/v0.5.3-rtd/concepts/environments.html

For dataset download, the old urls are the correct ones. The new ones which are mostly hosted on github or hugging face are for maniskill 3. Unfortunately download speed is heavily dependent on your network and our servers and can be variable.

yingfei1016, 2024-07-25:
Thank you very much! It helps me o lot!

closed 2024-07-25 by yingfei1016