mani-skill/ManiSkill #457: Where is the user_solution.py for the evalution?

yingfei1016, 2024-07-27:
Hi,

I following the old document to process the data and train my model. After that I want to test my model locally, but I find the template code (https://github.com/haosulab/ManiSkill2/tree/main/examples/submission) is out-of-date, where can I find this code?

[image]

StoneT2000, 2024-07-30:
I think it refers to https://github.com/haosulab/ManiSkill/tree/v0.5.3/mani_skill2/evaluation/solution.py

However this is just for online evaluation, I don't think we are running an online evaluation service anymore.

yingfei1016, 2024-07-30:
Thank you! I find the code from https://github.com/haosulab/ManiSkill2-Learn/blob/xmate-uncolored/submission_example/user_solution.py

closed 2024-07-30 by StoneT2000