mani-skill/ManiSkill #458: How to set the velocity limit of a joint?

hesic73, 2024-07-27:
In a URDF file, a joint can have <limit> elements (required for revolute and prismatic joints) with the following attributes:

> lower (optional, defaults to 0)
> 
> An attribute specifying the lower joint limit (in radians for revolute joints, in metres for prismatic joints). Omit if joint is continuous.
> 
> upper (optional, defaults to 0)
> 
> An attribute specifying the upper joint limit (in radians for revolute joints, in metres for prismatic joints). Omit if joint is continuous.
> 
> effort (required)
> 
> An attribute for enforcing the maximum joint effort (|applied effort| < |effort|). [See safety limits](https://wiki.ros.org/pr2_controller_manager/safety_limits).
> 
> velocity (required)
> 
> An attribute for enforcing the maximum joint velocity (in radians per second [rad/s] for revolute joints, in metres per second [m/s] for prismatic joints). [See safety limits](https://wiki.ros.org/pr2_controller_manager/safety_limits).

However, it appears that the ArticulationJoint class in the [ManiSkill repository](https://github.com/haosulab/ManiSkill/blob/1c64f1114b72840800b02268661f6353a003b889/mani_skill/utils/structs/articulation_joint.py#L25) does not provide a method to handle the `velocity` attribute.

StoneT2000, 2024-07-28:
As far as I know this does not exist.

if you are doing motion planning, it is better to set velocity limits through your motion planner. If you do RL, penalize the velocities. The other way is to modify the PD parameters of the controllers in the code or force your agent to take smaller actions

hesic73, 2024-07-28:
No problem, I was just curious about it. I'm looking into implementing something similar to an IsaacGym project where velocity settings are available. Thank you for the clarification!

closed 2024-07-28 by hesic73