mani-skill/ManiSkill #460: Running on Multi-GPUs

HenryWJL, 2024-07-29:
Hi, I want to run ManiSkill on multiple GPUs, but every time I tried, only 'cuda:0' was being used. How can I enable other GPUs?

StoneT2000, 2024-07-29:
We just added support for picking which device to use in the latest commit of this github repo. The pypi version hasn't been updated just yet. I will be updating it in a few days after I close a few more issues/bugs.

StoneT2000, 2024-07-29:
See my response earlier here: https://github.com/haosulab/ManiSkill/issues/447 for how to set it up if you don't want to wait for pypi upload.

HenryWJL, 2024-07-29:
Thank you very much! I will take a try.

HeegerGao, 2024-07-29:
Hi, do you have example codes for using multi-GPU for simulation (I mean multiprocessing on ManiSkill envs with GPU simulation). Thanks!

StoneT2000, 2024-07-29:
Not yet sorry @HeegerGao, what is your use case in particular? I haven't prioritized trying it out since most use cases at the moment can be tackled on a single GPU easily.

HeegerGao, 2024-07-30:
Thanks for your prompt reply! I am training a dexterous manipulation task with reinforcement learning and using point cloud as input. Currently I find that one A100 GPU can support about 120 parallel envs. We will see if this is enough, but we would appreciate it if you could provide an official multi-processing example.

HenryWJL, 2024-07-30:
Hi, I followed your instructions and installed the latest version. Then, I set `sim_backend="cuda:1"` and the following error occurred:

`Traceback (most recent call last):
  File "/data/wjl/miniconda3/envs/contact_realization/lib/python3.9/site-packages/torch/multiprocessing/spawn.py", line 68, in _wrap
    fn(i, *args)
  File "/data/wjl/projects/contact_realization/train.py", line 79, in run
    envs = ManiSkillEnv(
  File "/data/wjl/projects/contact_realization/contact_realization/env/mani_skill.py", line 59, in __init__
    envs = gym.make(
  File "/data/wjl/miniconda3/envs/contact_realization/lib/python3.9/site-packages/gymnasium/envs/registration.py", line 802, in make
    env = env_creator(**env_spec_kwargs)
  File "/data/wjl/projects/contact_realization/mani_skill/utils/registration.py", line 167, in make
    env = env_spec.make(**kwargs)
  File "/data/wjl/projects/contact_realization/mani_skill/utils/registration.py", line 78, in make
    return self.cls(**_kwargs)
  File "/data/wjl/projects/contact_realization/mani_skill/envs/tasks/contact/realize_contact.py", line 116, in __init__
    super().__init__(
  File "/data/wjl/projects/contact_realization/mani_skill/envs/sapien_env.py", line 313, in __init__
    obs, _ = self.reset(seed=2022, options=dict(reconfigure=True))
  File "/data/wjl/projects/contact_realization/mani_skill/envs/sapien_env.py", line 721, in reset
    self._reconfigure(options)
  File "/data/wjl/projects/contact_realization/mani_skill/envs/sapien_env.py", line 588, in _reconfigure
    self.scene._setup_gpu()
  File "/data/wjl/projects/contact_realization/mani_skill/envs/scene.py", line 696, in _setup_gpu
    actor.set_pose(actor.initial_pose)
  File "/data/wjl/projects/contact_realization/mani_skill/utils/structs/actor.py", line 362, in set_pose
    self.pose = arg1
  File "/data/wjl/projects/contact_realization/mani_skill/utils/structs/actor.py", line 346, in pose
    self._body_data_index[self.scene._reset_mask[self._scene_idxs]], :7
RuntimeError: indices should be either on cpu or on the same device as the indexed tensor (cuda:1)`

How can I fix this error?

chenyenru, 2024-07-30:
> Hi, I followed your instructions and installed the latest version. Then, I set `sim_backend="cuda:1"` and the following error occurred:
>
>
>
> `Traceback (most recent call last):
>
>   File [...]

Hey could you share how the environment was initialized. Seems like some tensors are not on cuda:1. Was `render_backend="cuda"` specified per the instruction [here](https://maniskill.readthedocs.io/en/latest/user_guide/getting_started/quickstart.html#additional-gpu-simulation-rendering-customization)?

HenryWJL, 2024-07-31:
```python
import gymnasium as gym
import mani_skill.envs

env = gym.make(
    "PickCube-v1",
    num_envs=16,
    sim_backend="cuda:1",
    render_backend="cuda",
)
```
Hi, I ran the test script above and the same error was raised. Seems like something was wrong with the base environment.

StoneT2000, 2024-07-31:
@HenryWJL Found the bug. There are a few places form our old code that assumed default cuda:0 as the device. will try and fix soon.

HenryWJL, 2024-07-31:
OK, I will be waiting.

StoneT2000, 2024-07-31:
Ok the fix is actually quite simple, you shouldn't set the sim or render backends actually, instead you should export the visible cuda devices. Pull the latest git commit, then before running your python code, run
```
export CUDA_VISIBLE_DEVICES=1
```
which select the GPU device 1 (you can change this to any number. Then create environments like normal, without specifying sim or render backends.

Currently there may be some rendering bugs if you try to e.g. `export CUDA_VISIBLE_DEVICES=0,1` expose multiple GPUs to a single python process and select the render backend to be anything but the first exposed GPU. But for most use cases and eventually a multiprocessing setup (an example I am still working) will not need to have multiple cuda devices exposed to a single python process.

StoneT2000, 2024-07-31:
by the way @HeegerGao I just remembered, with a multiprocessing setup an issue is the rendered PCDs/visual data of each python subprocess will be on different GPUs. There can be some slowness in transfering data from one GPU to another as a result (and off the top of my head i am not even sure how to send cuda tensors via multiprocessing to a main process, there probably is a way). Hence I would recommend adopting a distributed learning approach. E.g. PPO used to be implemented in a way where it would run N sub-processes to collect data and generate gradients, then average the gradients in the main process before updating.

StoneT2000, 2024-07-31:
https://isaac-sim.github.io/IsaacLab/source/features/multi_gpu.html#multi-gpu-training sums it up nicely in a figure of how that works. We don't have an example of it with torch distributed but I imagine that is the next step for us to investigate.

HeegerGao, 2024-07-31:
Thanks for your explanation! Yes, I understand it is difficult to transfer cuda tensors across different GPUTs during the multiprocessing. Currently, I and @HenryWJL are trying to use the traditional CPU-based PPO multiprocessing implementation (like done in [ManiSkill2-Learn](https://github.com/haosulab/ManiSkill2-Learn)): we try to use N sub-process for N GPUs, and in each sub-process it has around 120 GPU-based parallel envs. Each sub-process (i.e., GPU) has one copy of the network, and it will explore the env and compute gradients separately, finally, the main subprocess averages the gradients from N*120 envs to update the network. Is this correct?

HenryWJL, 2024-07-31:
> Ok the fix is actually quite simple, you shouldn't set the sim or render backends actually, instead you should export the visible cuda devices. Pull the latest git commit, then before running your python code, run
> 
> ```
> export CUDA_VISIBLE_DEVICES=1
> ```
> 
> which select the GPU device 1 (you can change this to any number). Then create environments like normal, without specifying sim or render backends.
> 
> Currently there may be some rendering bugs if you try to e.g. `export CUDA_VISIBLE_DEVICES=0,1` expose multiple GPUs to a single python process and select the render backend to be anything but the first exposed GPU. But for most use cases and eventually a multiprocessing setup (an example I am still working) will not need to have multiple cuda devices exposed to a single python process.
> 
> What effectively is happening is torch/our simulator thinks "cuda:0" is the GPU it is using while actually we only let it use gpu 1, not 0.

Thank you for your comments, the problem is now solved.

closed 2024-07-31 by StoneT2000

Anonymous-N, 2025-07-16:
> Thanks for your explanation! Yes, I understand it is difficult to transfer cuda tensors across different GPUTs during the multiprocessing. Currently, I and [@HenryWJL](https://github.com/HenryWJL) [...]

Hi there, have you already implemented this feature?  
If so, could you please give me some hints or share a small example? Thanks!

youwyu, 2025-09-21:
> import gymnasium as gym
> import mani_skill.envs
> 
> env = gym.make(
>     "PickCube-v1",
>     num_envs=16,
>     sim_backend="cuda:1",
>     render_backend="cuda",
> )
> Hi, I ran the test script above and the same error was raised. Seems like something was wrong with the base environment.

I also simply tried to setup one device for Physx and one device for Sapien, and it failed. My issue is not about "indices should be either on cpu or on the same device as the indexed tensor", since I have modified the source code to map the indices.
Could the author please provide some updates or hints?

StoneT2000, 2025-09-21:
The backend has to be the same, you can't dedicate one GPU for rendering one for simulation.