mani-skill/ManiSkill #469: Question about data manage while I create 6 fake joints to control floating joints

SparKgod1, 2024-08-01:
Yep you exactly do that. Create 6 fake joints to control XYZ RPY in the URDF as parents of the base link.. Any order should work. I can publish an official floating panda gripper example next week.

_Originally posted by @StoneT2000 in https://github.com/haosulab/ManiSkill/issues/406#issuecomment-2212353677_

SparKgod1, 2024-08-01:
According to [this](https://maniskill.readthedocs.io/en/latest/user_guide/concepts/gpu_simulation.html#data-organization-on-the-gpu),the data must  include pose (7D), linear velocity (3D), and angular velocity (3D) for every rigid body actor and articulation link ,but I add 6 fake joints to control XYZ RPY in the URDF.Do I need to repeat the XYZ RPY in pose (7D)?

SparKgod1, 2024-08-01:
I didn't describe the problem very well earlier. Based on the code for setting the state below, should I input the pose in both set_root_pose and set_qpos, or should I choose only one of them?
https://github.com/haosulab/ManiSkill/blob/825dc6ce094ed072e1574954645fb6bbf008153a/mani_skill/utils/structs/articulation.py#L258-L281

SparKgod1, 2024-08-01:
If I set the pose in both set_root_pose and set_qpos, it will look like this:

https://github.com/user-attachments/assets/8b253b34-39cd-473f-ad24-a7e8b5b9314f

However, if I set the pose only in set_root_pose, the problem seems to be solved.
But it seems that the joint positions are not set correctly. When I apply each frame of data directly to the allegro hand , I get the correct pose, but using replay results in an incorrect finger pose.

my replay command:

        python -m mani_skill.trajectory.replay_trajectory --traj-path my_traj_data/trajectory_right.h5 \
            -c pd_joint_pos -o state \
            --save-traj --count 1 \
            --record-rewards true \
            --allow-failure \
            --use-env-states \
            --save-video

and I set the control mode to pd_joint_pos in the json file of trajectory,I wonder if it might be an issue with the control mode.I would sincerely appreciate your guidance, thank you.

StoneT2000, 2024-08-01:
Joints let you control the robot hand in air freely. There are 6 dimensions because in the real world thats how many dimensions there are. Pose is 7D because we use the quaternion representation which doesn't have an easy (and thus teloperatable) visualization. 

Since you are trying to control the hand. the root pose in fact should just be fixed at 0, and you can just set the initial joint qpos values of the first 6 joints for your starting point. This way, the robot's qpos[:3] (first 3 qpos) are your robot's XYZ position in the world. Naturally qpos[3:6] is then your robots RPY values. This is sufficient, no need to change the stored robot pose.

SparKgod1, 2024-08-01:
> Joints let you control the robot hand in air freely. There are 6 dimensions because in the real world thats how many dimensions there are. Pose is 7D because we use the quaternion representation [...]

@StoneT2000 Thank you for your response. I have another question. I followed your advice and fixed the root node at 0. When I use `python test.py -r "allegro_hand_right" -k "grasping_pose"` I found that the grasping pose in every frame is all normal, as shown in the picture below:

[image]

However, the trajectory reproduced using replay is still chaotic. Why is this happening? I hope you can give me some guidance. I would be very grateful!

SparKgod1, 2024-08-01:
Oh, I also input qpos in the action. Could this be the reason for the problem?

SparKgod1, 2024-08-01:
But setting all actions to zero doesn't seem to solve the problem either. Could you give me some advice on how to organize my own dataset to train (RFCL)[https://github.com/haosulab/ManiSkill/tree/main/examples/baselines/rfcl#download-and-process-dataset]?

closed 2024-08-01 by SparKgod1

StoneT2000, 2024-08-01:
Oh did you resolve this issue already? @SparKgod1

SparKgod1, 2024-08-02:
> Oh did you resolve this issue already? @SparKgod1

@StoneT2000 Ah,yes, I found out that I got the joint positions wrong in the controller, haha. But now I've encountered a new problem: the joints can't reach the positions specified by my action, and some joints even move in the opposite direction of the set target. After using use_env_state to correct it, there is still a significant deviation from the target. The replay result is as follows:

https://github.com/user-attachments/assets/fea824dd-9ea0-4767-b242-44fe4fcdcb30

It should be like this:

[origin data retargeting result but no collision](https://github.com/dexsuite/dex-retargeting/tree/main/example/position_retargeting#retarget-robot-motion-from-hand-object-pose-dataset)

Below are the qpos values for one frame that I printed out. The first six are XYZ and RPY, and the remaining sixteen are joint rotations. It can be seen that the positional control is relatively accurate, with only slight deviations caused by the controller. However, the errors in some joints are very large.

[image]

I now suspect that the large errors are due to self-collision of the hand joints, preventing the joints from reaching the specified positions. This is because when I cancel the use_env_state, the incorrect actions cannot be corrected, and the hand's qpos directly becomes NaN. It seems that only improper collisions would cause this. Additionally, when visualizing the hand's posture using keyframe, I noticed that some joints overlap during grasping (as shown in the figure below). Of course, the slight deviations in some normally functioning joints are likely due to the precision of the controller.

[image]

Since I'm retargeting human hand data to the Allegro hand, I think this might be causing joint overlap. Therefore, I'm considering whether I should switch to a five-finger hand model, such as the SVH, as my next step. Could this solve the joint position errors? Could you give me some advice? Thank you very much.

SparKgod1, 2024-08-03:
> > Oh did you resolve this issue already? @SparKgod1
> 
> @StoneT2000 Ah,yes, I found out that I got the joint name order wrong in the controller, haha. But now I've encountered a new problem: the joints can't reach the positions specified by my action, and some joints even move in the opposite direction of the set target. After using use_env_state to correct it, there is still a significant deviation from the target. The replay result is as follows:
> 0.mp4
> 
> It should be like this:
> 
> [origin data retargeting result but no collision](https://github.com/dexsuite/dex-retargeting/tree/main/example/position_retargeting#retarget-robot-motion-from-hand-object-pose-dataset)
> 
> Below are the qpos values for one frame that I printed out. The first six are XYZ and RPY, and the remaining sixteen are joint rotations. It can be seen that the positional control is relatively accurate, with only slight deviations caused by the controller. However, the errors in some joints are very large.
> 
> ![微信图片_20240802093553](https://private-user-images.githubusercontent.com/131428186/354429757-0cb31351-e41c-4f89-8ce9-70c416d4ca7b.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjI2Nzg1NzgsIm5iZiI6MTcyMjY3ODI3OCwicGF0aCI6Ii8xMzE0MjgxODYvMzU0NDI5NzU3LTBjYjMxMzUxLWU0MWMtNGY4OS04Y2U5LTcwYzQxNmQ0Y2E3Yi5qcGc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwODAzJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDgwM1QwOTQ0MzhaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT00MGRkYWRjMTc2M2I2ZGQxZDcxMDAzYzQ4NzY0NTZiZWZjZjZlYWRmOTdiOGE2MGI1ZWVmNGRkNmQ3ZTVhOTc0JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.DLC9b-TSfmzrZK13dY-3-R9h9ce6T95pp1yMcQhDsYw)
> 
> I now suspect that the large errors are due to self-collision of the hand joints, preventing the joints from reaching the specified positions. This is because when I cancel the use_env_state, the incorrect actions cannot be corrected, and the hand's qpos directly becomes NaN. It seems that only improper collisions would cause this. Additionally, when visualizing the hand's posture using keyframe, I noticed that some joints overlap during grasping (as shown in the figure below). Of course, the slight deviations in some normally functioning joints are likely due to the precision of the controller.
> 
> ![compressed_image](https://private-user-images.githubusercontent.com/131428186/354429466-3fae0435-bcd6-48fb-83aa-72a758c084ab.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjI2Nzg1NzgsIm5iZiI6MTcyMjY3ODI3OCwicGF0aCI6Ii8xMzE0MjgxODYvMzU0NDI5NDY2LTNmYWUwNDM1LWJjZDYtNDhmYi04M2FhLTcyYTc1OGMwODRhYi5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwODAzJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDgwM1QwOTQ0MzhaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT00YzllYjAyZGJhMDFhZDA0NDVhMmQyYTJkNjA4YTY1NDY1ZWU0MjEyN2VjNThkY2I1N2JlYTI4NTJhOWY4OTg4JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.kpmfumhoMFwVFnlE23MXUgCG2Ji3RaLjsqdXtRsiyrs)
> 
> Since I'm retargeting human hand data to the Allegro hand, I think this might be causing joint overlap. Therefore, I'm considering whether I should switch to a five-finger hand model, such as the SVH, as my next step. Could this solve the joint position errors? Could you give me some advice? Thank you very much.

I solved the problem by switching to SVH! Thank you for your patient explanation. This work is very valuable, and I hope future updates will be even more exciting. Thanks!

SparKgod1, 2024-08-03:
@StoneT2000 I solved the problem by switching to SVH! Thank you for your patient explanation. This work is very valuable, and I hope future updates will be even more exciting. Thanks!