mani-skill/ManiSkill #471: TwoRobotStackCube Task

Nielsencu, 2024-08-01:
Thank you for this awesome project and sharing the codebase! I went to test out the TwoRobotStackCube task using the PPO algorithm, and after the training ended, the model did not converge (0% success rate). Can I check if this particular task has been tested to work with any of the baselines training algorithm? In the paper I did read about the StackCube task being tested, but not much on TwoRobotStackCube. Hope you can shed some light on this! Thank you

StoneT2000, 2024-08-01:
TwoRobotStackCube's dense reward isn't very super optimal (hence I don't include an example script of it in the README). A stronger RL algorithm might be able to solve it but I haven't tested it. I think I got PPO to work once but it was a lucky run, you need to tune the hyperparameters.

Nielsencu, 2024-08-01:
I see. Thank you for your prompt response!

closed 2024-08-01 by Nielsencu