mani-skill/ManiSkill #472: There is a discrepancy of about 2 cm between the TCP pose planned by mplib and the final TCP pose read from the Franka robot

qiaojunyu, 2024-08-01:
Great work and codebase, and I have developed something interesting using this simulator. When we use mplib to plan the trajectory of the Franka robot in the built-in environment, we input the target TCP pose, but find that the final TCP pose differs from the specified TCP pose by about 2 cm.

TianxingChen, 2024-08-03:
I met the same error. It's variable. I don't know how to reduce this error, such as changing a specific threshold.

StoneT2000, 2024-08-04:
It's hard to tell what the error is without it an example. If you are using the example motion planning code my guess is you might be tracking the wrong link. The panda arm has a invisible link called tcp or something which is a few centimeters below the end effector link

StoneT2000, 2024-08-09:
Any update @qiaojunyu @TianxingChen for some example code to test to reproduce the problem?

closed 2024-08-30 by StoneT2000