mani-skill/ManiSkill #477: Controller tuning for custom robot

Nielsencu, 2024-08-04:
I managed to import a Kinova Gen3 robot arm with the Robotiq 2F-85 into the simulation. I also followed the custom robots guide stated below 

https://maniskill.readthedocs.io/en/latest/user_guide/tutorials/custom_robots.html

I've verified that my robot with pd_joint_delta_pos controller and random action moves similar to the video shown and also the pd_joint_pos controller maintains the rest keyframe accordingly.

However, when I trained the robot on StackCube-v1 task it approaches the cube but doesn't really grip it and it behaves quite weirdly.

I have attached a video showing the training (step 0 and step 104).

https://github.com/user-attachments/assets/298341ff-dc29-41ef-aa58-982bd7141346

https://github.com/user-attachments/assets/0b17e191-ee85-4d0e-8c85-230bd5c56f21

Do you have some other tips for tuning the P, D, and also force limit values for the arm and the gripper that is more effective e.g tuning the PD values real-time with the simulator interactively? I also realized for all grippers, such as fetch and panda they have the same PD and force limit coefficients, does that mean for my robot it should also be the same?

StoneT2000, 2024-08-06:
the P, D and force limit values in simulation don't matter that much, you can probably just copy the values for fetch/panda for your implementation. They only matter if you want to do sim2real transfer then you need to system ID by checking if an action in sim corresponds to the same behavior in the real world.

I assume you are using PPO to train. Maybe first start with the PushCube and PickCube tasks. StackCube is actually much harder than it looks (most RL algorithms solve this task slower than most existing robot RL benchmarks).

Another thing, it looks like the robot is initialized at the middle of the table. This might make the task too hard to solve. If you are using the stack cube reward function, it encourage the gripper to move towards the red cube first, then grasp it. However the current position of the robot might make this too hard since the cube is often very close to the robot base.

Another thing is video 3.mp4 the full video? As in at the start of the evaluation episode it looks like that?

Nielsencu, 2024-08-12:
Thank you for your suggestion. Indeed, I realized that the PD values are not the issue.

I finally found the culprit and it is due to the end effector (Robotiq-2F-85), it has a peculiar configuration, having 6 actively controlled joints compared to the conventional 2-joint EE such as one's used for Panda.

In Gazebo, I used the joint state publisher gui to control the gripper for open and closing and it works fine. Video below shows manually moving just the finger_joint through the GUI.

https://github.com/user-attachments/assets/426692cb-7e39-48c2-a8a1-18c8ac05cbd5

However, when exporting it from .xacro to urdf and loading the URDF into Maniskill, it behaves weirdly. Video below is again manually moving the finger_joint through the slider, but robot goes haywire.

https://github.com/user-attachments/assets/78010224-d75a-4a9a-85fd-bd974dc9eadd

Note that the finger_joint is a mimic_joint, which moves other joints that have a mimic tag referencing the finger_joint. Perhaps this is what's causing the problem? Here's the URDF if it's helpful to debug.
[gen3_robotiq_2f_85_ori.zip](https://github.com/user-attachments/files/16587879/gen3_robotiq_2f_85_ori.zip)

One thing though the robotiq (2f-85) gripper was used in ManiSkill few years back, and I compared my URDF with the one in the repo below:

https://github.com/haosulab/cvpr-tutorial-2022/blob/master/assets/robotiq_2f_85_gripper_visualization/robotiq_85_original.urdf

It's also seem to be using mimic joint, but all the inertial component in the links got removed. I'm afraid disabling this might not translate well for sim-to-real.

Nonetheless,I tried removing the inertial links from the robot, however, still no good results (shown below).

https://github.com/user-attachments/assets/bf251d69-6865-4bb9-9498-aee3be8a56f0

When all joints in the gripper were disabled, the robot could move normally.

https://github.com/user-attachments/assets/cb53c8a2-2ce9-463c-a468-c777c3ea4429

I used Gen3 with all gripper joints disabled and tested it on the PushCube task and initialized the robot closer to the edge of the table, following your suggestion, and it can learn the task by using its arm to push the cube to target destination although not 100% success rate after 10 million steps.

Hope you can give some advice regarding this! Let me know if you need more information, and thank you so much for the help.

Nielsencu, 2024-08-16:
Just wanted to check if there's any suggestions regarding this issue, would be greatly appreciated! :)

StoneT2000, 2024-08-16:
Will bump this up my TODO list today, thanks for the reminder

StoneT2000, 2024-08-19:
For your URDF you will also need to send me the full folder, I am missing the mesh files

Nielsencu, 2024-08-20:
Meshes: [kortex_description.zip](https://github.com/user-attachments/files/16669269/kortex_description.zip)
URDF: 
[kinovagen3.zip](https://github.com/user-attachments/files/16669283/kinovagen3.zip)

UltronAI, 2024-08-20:
Hi @StoneT2000,

I'm working on a similar project and encountered the same issues—specifically, the arm motion being affected by changes in the gripper joints. I also struggled to apply the correct constraints to the Robotiq gripper.

I followed the tutorial linked in [this comment](https://github.com/haosulab/ManiSkill/issues/71#issuecomment-1476948160) and implemented [debug/robotiq.py](https://github.com/haosulab/cvpr-tutorial-2022/blob/master/debug/robotiq.py) using the latest versions of SAPIEN 3. While some behaviors are a bit unusual, it seems mostly correct.

[Screencast from 08-20-2024 06:11:51 PM.webm](https://github.com/user-attachments/assets/21f22132-d0b5-4387-994d-4ee3e7088dbc)

However, when I used this approach to build a tabletop environment with the Robotiq gripper in ManiSkill 3, I encountered an unexpected issue. Given the success with SAPIEN alone, I'm unsure what’s causing the problem.

[Screencast from 08-20-2024 06:17:51 PM.webm](https://github.com/user-attachments/assets/14eb3e85-ccaf-4f29-9c31-436c02acf31d)

To reproduce this issue, you can use the code in [this fork](https://github.com/UltronAI/ManiSkill/tree/debug_robotiq).

`mani_skill/examples/debug_robotiq_sapien.py` contains my reimplementation of the tutorial code. Run it with `python -m mani_skill.examples.debug_robotiq_sapien.py --model_constraint`.

Running `python -m mani_skill.examples.debug_robotiq_maniskill.py` will reproduce my results in ManiSkill 3. The corresponding Robotiq agent is in [mani_skill/agents/robots/robotiq/robotiq.py](https://github.com/UltronAI/ManiSkill/blob/debug_robotiq/mani_skill/agents/robots/robotiq/robotiq.py). Unfortunately, I cannot control the gripper correctly with either the PDJointPosMimicController or by directly changing its joint positions via GUI.

Thank you in advance for any advice on resolving this issue!

StoneT2000, 2024-08-20:
Ok one issue is the old cvpr tutorial urdf definition is actually outdated/partly wrong. I am fixing it to make it work. 

Also after discussing with @fbxiang, to simulate this very specific gripper well/accurately, it requires some complex drive creation code so mimic joints labels in the urdf cannot really be used. I'm hoping we can think of some general way to define complex grippers like this one via drives (which simulate better and are more stable).

StoneT2000, 2024-08-20:
Fixed URDF files: https://github.com/haosulab/ManiSkill-Robotiq_2F/tree/v0.1.0 (there's a floating one and a non floating one, feel free to add these to your own robots). Also fixes the visual meshes so its not all gray.

@Nielsencu you should be able to look at the gripper implementation and copy it over to your own custom robot with the kinova setup:

StoneT2000, 2024-08-20:
Fixed URDF files: https://github.com/haosulab/ManiSkill-Robotiq_2F/tree/v0.1.0 (there's a floating one and a non floating one, feel free to add these to your own robots). Also fixes the visual meshes so its not all gray.

@Nielsencu you should be able to look at the floating robot iq 2f 85 implementation and copy it over to your own custom robot with the kinova setup. Key thing to do is to add the correct passive/mimic controller configs and define _after_load_articulation function to setup the drives which make the gripper behave like real life.

@UltronAI You can try out the new floating gripper
```bash
python mani_skill/examples/demo_robot.py -r "floating_robotiq_2f_85_gripper" --zero-actions -c pd_joint_delta_pos
```
The drives are setup so only the outer knuckle joints actually control the gripper, the rest are passive. This will open the GUI and you can click the articulation, then the outer knuckle joint and set the position target to see it.
[image]

closed 2024-08-30 by StoneT2000

AcademeMaster, 2024-11-30:
[2024年11月30日 屏幕视频 21时15分13秒.webm](https://github.com/user-attachments/assets/feaa3f31-50b4-4c95-8901-76b5bc085f9d)
hello,I use this urdf and  floating_robotiq_2f_85 congtorller ，、but it's not work

AcademeMaster, 2024-11-30:
this is my code:

    from typing import Dict, Union
    import numpy as np
    import sapien
    
    from mani_skill import ASSET_DIR, PACKAGE_ASSET_DIR
    from mani_skill.agents.base_agent import BaseAgent, DictControllerConfig, Keyframe
    from mani_skill.agents.controllers import *
    from mani_skill.agents.controllers.base_controller import ControllerConfig
    from mani_skill.agents.registration import register_agent
    from mani_skill.sensors.camera import CameraConfig
    from mani_skill.utils import sapien_utils
    
    
    @register_agent(asset_download_ids=["robotiq_2f"])
    class FloatingRobotiq2F85Gripper(BaseAgent):
        uid = "floating_robotiq_2f_85_gripper"
        urdf_path = f"{ASSET_DIR}/robots/robotiq_2f/floating_robotiq_2f_85.urdf"
        disable_self_collisions = True
        urdf_config = dict(
            _materials=dict(
                gripper=dict(static_friction=2.0, dynamic_friction=2.0, restitution=0.0)
            ),
            link=dict(
                left_inner_finger_pad=dict(
                    material="gripper", patch_radius=0.1, min_patch_radius=0.1
                ),
                right_inner_finger_pad=dict(
                    material="gripper", patch_radius=0.1, min_patch_radius=0.1
                ),
            ),
        )
        keyframes = dict(
            open_facing_up=Keyframe(
                qpos=[0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0],
                pose=sapien.Pose(p=np.array([0.0, 0.0, 0.5])),
            ),
            open_facing_side=Keyframe(
                qpos=[0, 0, 0, 0, 0, 0],
                pose=sapien.Pose(
                    p=np.array([0.0, 0.0, 0.5]), q=np.array([0.7071, 0, 0.7071, 0])
                ),
            ),
        )
        root_joint_names = [
            "root_x_axis_joint",
            "root_y_axis_joint",
            "root_z_axis_joint",
            "root_x_rot_joint",
            "root_y_rot_joint",
            "root_z_rot_joint",
        ]
    
        @property
        def _controller_configs(
            self,
        ) -> Dict[str, Union[ControllerConfig, DictControllerConfig]]:
    
            # define a simple controller to control the floating base with XYZ/RPY control.
            base_pd_joint_pos = PDJointPosControllerConfig(
                joint_names=self.root_joint_names,
                lower=None,
                upper=None,
                stiffness=1e3,
                damping=1e2,
                force_limit=100,
                normalize_action=False,
            )
            base_pd_joint_delta_pos = PDJointPosControllerConfig(
[... 65 lines cut ...]
            # the next 4 magic arrays come from https://github.com/haosulab/cvpr-tutorial-2022/blob/master/debug/robotiq.py which was
            # used to precompute these poses for drive creation
            p_f_right = [-1.6048949e-08, 3.7600022e-02, 4.3000020e-02]
            p_p_right = [1.3578170e-09, -1.7901104e-02, 6.5159947e-03]
            p_f_left = [-1.8080145e-08, 3.7600014e-02, 4.2999994e-02]
            p_p_left = [-1.4041154e-08, -1.7901093e-02, 6.5159872e-03]
    
            right_drive = self.scene.create_drive(
                lif, sapien.Pose(p_f_right), pad, sapien.Pose(p_p_right)
            )
            right_drive.set_limit_x(0, 0)
            right_drive.set_limit_y(0, 0)
            right_drive.set_limit_z(0, 0)
    
            outer_finger = self.robot.active_joints_map["left_inner_finger_joint"]
            inner_knuckle = self.robot.active_joints_map["left_inner_knuckle_joint"]
            pad = outer_finger.get_child_link()
            lif = inner_knuckle.get_child_link()
    
            left_drive = self.scene.create_drive(
                lif, sapien.Pose(p_f_left), pad, sapien.Pose(p_p_left)
            )
            left_drive.set_limit_x(0, 0)
            left_drive.set_limit_y(0, 0)
            left_drive.set_limit_z(0, 0)

swuhk, 2025-02-19:
> [2024年11月30日 屏幕视频 21时15分13秒.webm](https://github.com/user-attachments/assets/feaa3f31-50b4-4c95-8901-76b5bc085f9d) hello,I use this urdf and floating_robotiq_2f_85 congtorller ，、but it's not work

By default, the pop-up sapien-viewer pauses simulation. You need to unclick "Pause" (usually on the right top corner).

PR #513 (2024-08-20): [Feature] Add floating robotiq_2f_85 gripper support + fix bugs with other robot arms that use that gripper.
Should close #477