mani-skill/ManiSkill #481: [BugFix] Add GPU IK solver fallback option

StoneT2000, 2024-08-05:
Currently IK based controllers like pd_ee_delta_pos/pose on GPU sim might be bugged due to the fast kinematics package which i used to compute jacobians for fast end effector control. A fall back option is the slower (but more stable) pytorch kinematics which is python torch based. Would require forking that repo since https://github.com/UM-ARM-Lab/pytorch_kinematics/issues/35 is not resolved yet.

siriusted, 2024-08-08:
hi, I've observed a hard to reproduce bug with GPU IK where sometimes the Jacobian returned contains zeros. is this similar to what you describe here and do you know what causes this?

StoneT2000, 2024-08-08:
Yep it is one of them. I will probably add the fallback option / temporarily switch out fast kinematics this week. At the moment only the CPU IK solver works correctly.

StoneT2000, 2024-08-09:
Feel free to try the latest commit now @siriusted @RomDeffayet

closed 2024-08-09 by StoneT2000

RomDeffayet, 2024-08-13:
Thanks, unfortunately I get a device error in pytorch kinematics now during initialization: 

rot_h = torch.cat((rot, zeros), dim=-2).reshape(-1, 4, 3)
RuntimeError: Expected all tensors to be on the same device, but found at least two devices, cuda:0 and cpu! (when checking argument for argument tensors in method wrapper_CUDA_cat)

It seems to come from the fact that the device argument is set to 'cpu' by default and is not overwritten in pytorch_kinematics.urdf._convert_transform

StoneT2000, 2024-08-13:
Can you share a reproducible script?

RomDeffayet, 2024-08-14:
`import torch
import mani_skill
import gymnasium

torch.set_default_device("cuda")
env = gymnasium.make("PushCube-v1", num_envs = 2, control_mode="pd_ee_delta_pose")`

It is easily fixed by adding `device="cpu"` in pytorch_kinematics.urdf, line 17. I opened an issue on PK's repo for that.

StoneT2000, 2024-08-15:
I opened a PR to fix that (plus another small bug) on their repo. For the time being I updated my own fork of it on pypi so you can now do pip install pytorch_kinematics==0.7.3

PR #485 (2024-08-09): [BugFix] Add back pytorch kinematics which is likely more stable
Fixes #481 and cleans up the IK solver code to be more organized.