mani-skill/ManiSkill #483: Computing end effector pose of robot

Strivers999, 2024-08-08:
Thank you for the excellent code. While using the arm in the simulator, I find it a bit confusing to calculate the robot arm's end effector pose from the robot's joint positions (qpos). Could you please explain how to do this efficiently?

[image]

Regarding the Fetch robot described in the repository, the observation data appears as shown in the attached image. How can I calculate the end effector pose using this information?

Thank you very much.

StoneT2000, 2024-08-08:
Either use a forward kinematic library or directly access link poses. The end-effector could be the robot link. By default without adding it yourself, maniskill does not output EE poses since we don't know which link is the link you might want. If you are building a custom task you have to add it in the `_get_obs_extra` function. 

For fetch specifically, the gripper/end-effector link is called "gripper_link" (you can check by using the GUI or inspecting the URDF). To get the pose in world coordinates, run `self.agent.robot.links_map["gripper_link"].pose`.

Strivers999, 2024-08-09:
Thanks for your quick reply~

closed 2024-08-09 by Strivers999