mani-skill/ManiSkill #498: Question on the effects of `use_target` in a controller config object

hesic73, 2024-08-15:
Hello! I've read the [documentation on Controllers](https://maniskill.readthedocs.io/en/latest/user_guide/concepts/controllers.html) and the [source code](https://github.com/haosulab/ManiSkill/blob/main/mani_skill/agents/controllers/pd_joint_pos.py). However, I still have some questions about the `use_target` property of a controller config object.

My current understanding is as follows:

- If `use_delta=False`, `use_target` does not alter the behavior of the controller.
- If `use_delta=True` and `use_target=False`, the controller behaves like a common delta controller, where the target qpos is set to current qpos + delta.

However, I am a bit confused about the behavior when `use_delta=True` and `use_target=True`. IInstead of using the current qpos, the controller uses `self._target_qpos`.  Although `PDJointPosController` has a `_target_qpos` attribute, it seems that this attribute is always initialized to the qpos on reset, which makes it unclear to me what purpose `use_target=True` serves in this context. Could you please explain how `use_target=True` is intended to be used? Thank you very much!

By the way, I noticed that `PDJointPosControllerConfig.use_delta` defaults to `False`, while `PDEEPosControllerConfig.use_delta` defaults to `True`. Is this difference in default values intentional?

hesic73, 2024-08-18:
Thank you for the clarification!  So when `use_delta=True` and `use_target=True`, the action is the delta of the target position.
Given this, if I were to implement something like DexPoint, where the end effector controller is used to control the arm to grasp an object, which mode  is recommended?

StoneT2000, 2024-08-19:
just use the default pd_ee_delta_pose, which is use_delta=True and use_Target=False. This is pretty common.