mani-skill/ManiSkill #502: Support systems without GPUs for cpu sim running only

StoneT2000, 2024-08-16:


junxnone, 2025-03-21:
is it works without Nvidia GPU now?

StoneT2000, 2025-03-21:
Actually you can run without nvidia GPU, you still need some gpu however (eg AMD)

Mac is also supported too if you have a working gpu

junxnone, 2025-03-21:
Thanks.

So it is not works if the platform with the Intel GPU only.

-----

I am working on porting the DL model to Intel GPU(OpenVINO), but looks like the maniskill looks for the cuda.

```
/python3.11/site-packages/mani_skill/examples/demo_random_action.py", line 83, in main
    env: BaseEnv = gym.make(
                   ^^^^^^^^^
/python3.11/site-packages/gymnasium/envs/registration.py", line 802, in make
    env = env_creator(**env_spec_kwargs)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
/python3.11/site-packages/mani_skill/utils/registration.py", line 182, in make
    env = env_spec.make(**kwargs)
          ^^^^^^^^^^^^^^^^^^^^^^^
/python3.11/site-packages/mani_skill/utils/registration.py", line 79, in make
    return self.cls(**_kwargs)
           ^^^^^^^^^^^^^^^^^^^
/python3.11/site-packages/mani_skill/envs/scenes/base_env.py", line 66, in __init__
    super().__init__(
/python3.11/site-packages/mani_skill/envs/sapien_env.py", line 231, in __init__
    self.backend = parse_sim_and_render_backend(sim_backend, render_backend)
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
/python3.11/site-packages/mani_skill/envs/utils/system/backend.py", line 60, in parse_sim_and_render_backend
    render_device = sapien.Device("cuda")
                    ^^^^^^^^^^^^^^^^^^^^^
RuntimeError: failed to find device "cuda"
```

StoneT2000, 2025-03-21:
as an argument for gym.make can you pass 

`gym.make(..., render_device="xpu")` maybe?

Otherwise I'm not sure what the right device is for intel GPUs, @fbxiang ?

fbxiang, 2025-03-21:
I would try without setting device (maybe setting it to "") when creating the renderer. If Intel GPU is the only GPU available, then that one should be picked (at least it is not a CPU). However, I cannot guarantee that all integrated GPUs can work. I have only tried some recent AMD and Intel ones.

closed 2025-05-13 by StoneT2000